cezanne/usbip-win #194: windows Client attach Linux Server(BSOD)

andy010106, 2020-08-24:
Hi cezanne,

I am using your code in windows10 as USBIP client(0.1.0)
Linux install from kernel for USBIP server .
Linux server bind USB device OK
Linux server listening OK
windows client attach face two random problem(Linux server usbipd.exe show device connect complete):
 1.BSOD
 2.after windows client attach for 5~10 seconds, it will stop attach event and close the usbip.exe.

Here is the BSOD Dump message:

Microsoft (R) Windows Debugger Version 10.0.19041.1 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Windows\Minidump\082020-23218-01.dmp]
Mini Kernel Dump File: Only registers and stack trace are available

Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 18362 MP (4 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Built by: 18362.1.amd64fre.19h1_release.190318-1202
Machine Name:
Kernel base = 0xfffff802`56000000 PsLoadedModuleList = 0xfffff802`564480f0
Debug session time: Thu Aug 20 03:41:43.399 2020 (UTC - 5:00)
System Uptime: 0 days 0:24:59.258
Loading Kernel Symbols
...............................................................
................................................................
................................................................
...............................
Loading User Symbols
Loading unloaded module list
.................
For analysis of this file, run !analyze -v
3: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

PAGE_FAULT_IN_NONPAGED_AREA (50)
Invalid system memory was referenced.  This cannot be protected by try-except.
Typically the address is just plain bad or it is pointing at freed memory.
Arguments:
Arg1: ffff96071c11d000, memory referenced.
Arg2: 0000000000000000, value 0 = read operation, 1 = write operation.
Arg3: fffff802722b4f2d, If non-zero, the instruction address which referenced the bad memory
	address.
Arg4: 0000000000000002, (reserved)

Debugging Details:
------------------

*** WARNING: Unable to verify timestamp for usbip_vhci.sys

Could not read faulting driver name

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.Sec
    Value: 7

    Key  : Analysis.DebugAnalysisProvider.CPP
    Value: Create: 8007007e on DESKTOP-GA1HIIK

    Key  : Analysis.DebugData
    Value: CreateObject

    Key  : Analysis.DebugModel
    Value: CreateObject

    Key  : Analysis.Elapsed.Sec
    Value: 46

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 99

    Key  : Analysis.System
    Value: CreateObject

BUGCHECK_CODE:  50

BUGCHECK_P1: ffff96071c11d000

BUGCHECK_P2: 0

BUGCHECK_P3: fffff802722b4f2d

BUGCHECK_P4: 2

READ_ADDRESS: fffff802565733b8: Unable to get MiVisibleState
Unable to get NonPagedPoolStart
Unable to get NonPagedPoolEnd
Unable to get PagedPoolStart
Unable to get PagedPoolEnd
fffff8025642a3b8: Unable to get Flags value from nt!KdVersionBlock
fffff8025642a3b8: Unable to get Flags value from nt!KdVersionBlock
unable to get nt!MmSpecialPagesInUse
 ffff96071c11d000 

MM_INTERNAL_CODE:  2

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

CUSTOMER_CRASH_COUNT:  1

PROCESS_NAME:  usbip.exe

TRAP_FRAME:  ffff888ce72783c0 -- (.trap 0xffff888ce72783c0)
NOTE: The trap frame does not contain all registers.
Some register values may be zeroed or incorrect.
rax=0000000000004000 rbx=0000000000000000 rcx=0000000000002cf0
rdx=0000000000000010 rsi=0000000000000000 rdi=0000000000000000
rip=fffff802722b4f2d rsp=ffff888ce7278550 rbp=0000000000000001
 r8=0000000000004000  r9=ffff96071c11bcf0 r10=0000000000000000
r11=ffff888ce7278688 r12=0000000000000000 r13=0000000000000000
r14=0000000000000000 r15=0000000000000000
iopl=0         nv up ei ng nz na po nc
usbip_vhci+0x4f2d:
fffff802`722b4f2d f3a4            rep movs byte ptr [rdi],byte ptr [rsi]
Resetting default scope

STACK_TEXT:  
ffff888c`e7278118 fffff802`561ecf1a : 00000000`00000050 ffff9607`1c11d000 00000000`00000000 ffff888c`e72783c0 : nt!KeBugCheckEx
ffff888c`e7278120 fffff802`5605875f : 0000005f`25f7ed30 00000000`00000000 00000000`00000000 ffff9607`1c11d000 : nt!MiSystemFault+0x1c5aba
ffff888c`e7278220 fffff802`561d041e : 00000000`00000000 00000000`00000000 00000000`00000001 00000000`00000000 : nt!MmAccessFault+0x34f
ffff888c`e72783c0 fffff802`722b4f2d : ffffa780`d28d2000 ffff9607`13d8b0d0 00000000`00000000 fffff802`722b246f : nt!KiPageFault+0x35e
ffff888c`e7278550 ffffa780`d28d2000 : ffff9607`13d8b0d0 00000000`00000000 fffff802`722b246f 00000000`00000000 : usbip_vhci+0x4f2d
ffff888c`e7278558 ffff9607`13d8b0d0 : 00000000`00000000 fffff802`722b246f 00000000`00000000 00000000`00000000 : 0xffffa780`d28d2000
ffff888c`e7278560 00000000`00000000 : fffff802`722b246f 00000000`00000000 00000000`00000000 ffffa780`d28d2000 : 0xffff9607`13d8b0d0

SYMBOL_NAME:  usbip_vhci+4f2d

MODULE_NAME: usbip_vhci

IMAGE_NAME:  usbip_vhci.sys

STACK_COMMAND:  .thread ; .cxr ; kb

BUCKET_ID_FUNC_OFFSET:  4f2d

FAILURE_BUCKET_ID:  AV_R_INVALID_usbip_vhci!unknown_function

OS_VERSION:  10.0.18362.1

BUILDLAB_STR:  19h1_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {1c5b19a2-ed00-161e-a44e-3791422c08ab}

Followup:     MachineOwner
---------

Wqrld, 2020-09-09:
Try using 3.0 UDE, that works fine for me on windows 10 18362 19h1

andy010106, 2020-09-10:
Thanks for your opinion.
After listening your opinion,I tried  3.0 UDE as Windows Client then attach again.
It appears another bug.
It seems it found the remote USB device,but Client can not recognize the USB device.
I have already plugin my USB to local client ,it can work normally.
I think it maybe not my windows Client cannot use my USB device.
But after attach it cannot recognize.
Here is the following pop up message:

USB device not recognize
The last USB you connected to your computer malfunctioned and Windows does not recognize it.
Recommendation
Try reconnecting the device.If Windows still does not recognize it,your device may not be working properly.

TheMohawkNinja, 2020-10-28:
Hello,

I am using the current 0.3.0 kmdf ude release and have also experienced a BSOD while attaching a keyboard and mouse from a remote Linux machine.

Please see below minidump.

===================================================================================

Microsoft (R) Windows Debugger Version 10.0.20153.1000 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Windows\Minidump\102820-8093-01.dmp]
Mini Kernel Dump File: Only registers and stack trace are available

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*
Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 18362 MP (16 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Edition build lab: 18362.1.amd64fre.19h1_release.190318-1202
Machine Name:
Kernel base = 0xfffff800`4d200000 PsLoadedModuleList = 0xfffff800`4d6461b0
Debug session time: Wed Oct 28 15:33:01.909 2020 (UTC - 5:00)
System Uptime: 0 days 6:12:04.885
Loading Kernel Symbols
...............................................................
................................................................
................................................................
................................
Loading User Symbols
Loading unloaded module list
............
For analysis of this file, run !analyze -v
nt!KeBugCheckEx:
fffff800`4d3c2ce0 48894c2408      mov     qword ptr [rsp+8],rcx ss:0018:ffff8789`ba763a90=000000000000000a
10: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

IRQL_NOT_LESS_OR_EQUAL (a)
An attempt was made to access a pageable (or completely invalid) address at an
interrupt request level (IRQL) that is too high.  This is usually
caused by drivers using improper addresses.
If a kernel debugger is available get the stack backtrace.
Arguments:
Arg1: ffffb404b122d388, memory referenced
Arg2: 0000000000000002, IRQL
Arg3: 0000000000000000, bitfield :
	bit 0 : value 0 = read operation, 1 = write operation
	bit 3 : value 0 = not an execute operation, 1 = execute operation (only on chips which support this level of status)
Arg4: fffff8004d260b62, address which referenced memory

Debugging Details:
------------------

*** WARNING: Unable to verify timestamp for udecx.sys
*** WARNING: Unable to verify checksum for win32k.sys

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 6405

    Key  : Analysis.DebugAnalysisProvider.CPP
    Value: Create: 8007007e on BENPC
[... 96 lines cut ...]

SYMBOL_NAME:  pmon+11b9

MODULE_NAME: pmon

IMAGE_NAME:  pmon.sys

STACK_COMMAND:  .thread ; .cxr ; kb

BUCKET_ID_FUNC_OFFSET:  11b9

FAILURE_BUCKET_ID:  AV_pmon!unknown_function

OS_VERSION:  10.0.18362.1

BUILDLAB_STR:  19h1_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {146d5e1c-c312-8f0e-6f37-5dc8aefa6e27}

Followup:     MachineOwner
---------

cezanne, 2021-01-12:
@andy010106 @TheMohawkNinja : Please retry with a new release [v0.3.2](https://github.com/cezanne/usbip-win/releases/tag/v0.3.2).

cezanne, 2021-01-23:
@andy010106 : I'll close this issue. Please reopen if you need.

closed 2021-01-23 by cezanne