cezanne/usbip-win #196: USB ID Database error

ProjectInitiative, 2020-09-08:
[image]

I have included an image, but on Windows 10 1909, I get a database error when trying to view all of the USB devices. I tried running the application in administrator mode as well and continue to get the same error.

I am not sure if this is a new Windows issue, or not as I am not fully familiar with how the code-base reads the USB manufacturer and product information. Any insight on this issue would be much appreciated!

ProjectInitiative, 2020-09-08:
Further information: the binary tested was v0.3.0, commit 96015c3

Wqrld, 2020-09-09:
@ProjectInitiative You need to download https://github.com/cezanne/usbip-win/blob/master/userspace/usb.ids into the folder with your .exe in order to have device identification, i agree that it should be a bit more clear in the error.

ProjectInitiative, 2020-09-10:
Ah, adding the usb.ids file seemed to solve the problem!

I think it would be helpful to update the error message. Or possible try and ping [here](http://www.linux-usb.org/usb.ids) if the file does not exist. I might look at opening a pull request in the future!

Thank you!

closed 2020-09-10 by ProjectInitiative