cezanne/usbip-win #201: Include usb id database / hwdata

septatrix, 2020-10-22:
Currently executing `usbip.exe list -l` fails to open the id database which is required to show the correct manufacturers and device names. It appears that the code searches for a database at the module path (whatever path that may be) but is unable to find one. I think these should be included or alternatively the error message should say where it expected the file to be so that one can manually download such a file from e.g. the [debian repository](https://packages.debian.org/sid/misc/hwdata) and place them at the correct location.

TheMohawkNinja, 2020-10-26:
I am also receiving this error and am furthermore unable to attach remote devices. Can you still attach remote devices, or are you getting an error like I am, and if so, is it the 0x6 error?

septatrix, 2020-10-26:
Everything else is working as expected. Probably best to open a separate issue for that.

spog, 2020-11-28:
Hi, i copied the 'userspace/usb.ids' from project sources to my path with 'usbip.exe'. Afterwards commands 'usbip.exe list -r <rem_ip>' and 'usbip.exe list -l' correctly display manufactures and device names.

septatrix, 2020-11-28:
I did not notice that they are included in this repo. However I still think they should somehow be included in the downloadable exe

KieranDevvs, 2020-11-29:
@septatrix Or just follow the instructions within the README.md file clearly. Its not embedded because you can update the data store without updating the binary.

septatrix, 2020-11-30:
> Or just follow the instructions within the README.md file clearly.

It is however not stated in the section for installing the _Windows USB/IP client_ (but it is stated for the server).

> Its not embedded because you can update the data store without updating the binary.

Okay I guess that makes sense. And as long as the driver is not signed the installation requires multiple manual steps anyhow so I do not see this as a problem yet. Should cezanne get the drivers signed in the future it would be nice to bundle everything in the exe IMO.

KieranDevvs, 2020-11-30:
@septatrix It doesn't need to be stated in that section because its not required for a client connection. The client just needs to connect to a remote TCP server and relay the data to the virtual USB device. The device list is only used on the server because that's where the device is going to be bound from.
If you want to use the data store on the client for whatever reason, then copy it, but that's not required for everyone else.

> it would be nice to bundle everything in the exe IMO.
I've already explained this, its not embedded because its independent from the binary. If a new device is added to the store, we don't want to recompile and release a new version of the software because nothing has changed other than the user friendly identification of a device.

cezanne, 2021-01-12:
@septatrix : As @KieranDevvs pointed out, an external usb.ids would suffice. usb.ids should be located at the same usbip.exe folder. As such, a more friendly manual is needed.

closed 2021-01-12 by cezanne