cezanne/usbip-win #202: attach_ude does not return

septatrix, 2020-10-22:
When running `usbip.exe attach_ude --remote=10.8.0.128 --busid=1-1.3` the command does not return.

I have not tried the non-ude version as that does not support my device but under linux the `attach` subcommand returns after calling. Running `attach_ude` on Windows on the other hand blocks the terminal. Nevertheless the device seems to get attached correctly and works without any problems.

amerkestijn, 2020-11-27:
I've got this working using the following repository: https://github.com/barbalion/usbip-win-client
Just use "Install.cmd" to set up the config & service. After you've done that use "Attach.cmd" to attach your configured device.

cezanne, 2021-01-23:
@septatrix : Blocking attach is one of the weaknesses of usbip-win. Non-blocking attach command will be started to develop.

cezanne, 2021-01-24:
@septatrix : Please check [dev branch](https://github.com/cezanne/usbip-win/tree/dev). It supports non-blocking attach.

septatrix, 2021-01-26:
I used the artifacts from https://ci.appveyor.com/project/cezanne/usbip-win/builds/37426780 and installed the ude version. Everything seemed to work as expected so from my side this issue is resolved (once it hits master).

<hr>

Some things I noticed in the instructions: The files are listed as `usbip_*_udf.*` whereas the files in the build contain `_ude.*`. Also you wrote that it will install both versions (wdm and ude) if the corresponding installation files are found - how would I then select which one to choose for attaching? I just installed the ude version because of that...

cezanne, 2021-01-26:
@septatrix : 
> Some things I noticed in the instructions: The files are listed as usbip_*_udf.* whereas the files in the build contain _ude.*. 

Oh.. typo.. thanks. I'll fix. 

> Also you wrote that it will install both versions (wdm and ude) if the corresponding installation files are found - how would I then select which one to choose for attaching? I just installed the ude version because of that...

`CMD> usbip.exe install -u`  I'll add this instruction to README.
You can see a simple help with `usbip.exe help install`.

septatrix, 2021-01-26:
> `CMD> usbip.exe install -u` I'll add this instruction to README.
> You can see a simple help with `usbip.exe help install`.

Thanks for the quick response. But if I were to install both drivers which one would `usbip.exe attach` use?

cezanne, 2021-01-27:
@septatrix : 
>  But if I were to install both drivers which one would usbip.exe attach use?

When no option is given, ude is tried first. So there's no way to install both drivers. I'll check if README is misleading in this regard.

cezanne, 2021-06-24:
@septatrix : This functionality was added since v0.3.4. This issue will be closed.

closed 2021-06-24 by cezanne

septatrix, 2021-06-24:
Yes of course. Thanks for working on this :D