cezanne/usbip-win #206: libusbip: error: udev_device_new_from_subsystem_sysname failed

Wangdaen, 2020-11-02:
在window中按照 https://github.com/cezanne/usbip-win 中提供的方法，进行 USB/IP for Windows 相关操作后后，以window10 作为USB/IP的服务端，WSL中Ubuntu子系统作为客户端，然后进行连接测试：
[image]
上图为window服务端绑定设备和开启服务的以及连接信息
[image]
上图为在WSL中Ubuntu子系统下连接服务时，抛出的错误信息
不知道是什么原因造成的，是按照该README.md进行相关的操作
window上usbip版本信息如下图：
[image]
Ubuntu系统中的配置信息：
[image]
无法确认libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
是由什么原因引起，可能是两个usbip版本不一致导致的吗？

Wangdaen, 2020-11-02:
[image]

Wangdaen, 2020-11-02:
@cezanne

519790441, 2020-11-03:
linux usbip版本太老，下载系统对应内核源码，在源码根目录/tools/usb/usbip里面编译用户态的可执行文件 usbip，用这个编好的可执行文件来继续映射看看

Wangdaen, 2020-11-03:
@cezanne 现在wsl当中的内核版本为4.19.104-microsoft-standard，下载window提供的内核源码进行编译依旧出现该错误，改用https://github.com/torvalds/linux.git 版本v5.9 ，linux最新内核版本重新编译usbip，依然出现该错误，没有找到原因

cezanne, 2020-12-26:
@Wangdaen @519790441 : Please, use english.

Wangdaen, 2020-12-29:
@cezanne  Is currently using version wsl2 Ubuntu 18.04.5 LTS, switch to https://github.com/torvalds/linux.git v5.9 version, the Linux kernel version latest recompiling usbip, still appear the error, didn't find the cause.
The initial estimate is that the version on the window is too old to connect to an existing mapping

cezanne, 2021-01-12:
@Wangdaen : You seemed not to load vhcd-hcd.ko module in WSL.

cezanne, 2021-01-26:
@Wangdaen : I'm going to close this issue. Please reopen if you want.

closed 2021-01-26 by cezanne