cezanne/usbip-win #207: Can't install 0.3.0 in Windows 10

Chris2011, 2020-11-08:
Hi,

I downloaded all the 7 files from the release of 0.3.0.
I can list the remote devices so I guess that the USBIP Server, which runs on my Raspi 3B+ is working correctly(?).

What doesn't work is to use usbip.exe install. I got this error:
```
usbip: error: usbip_vhci.inf or usbip_vhci.sys file not found
usbip: error: cannot install vhci driver package
```

Is the config still wrong and does it try to go to the old files? Because I have the new files usbip_vhci_ude.

I have WIndows 10 Home 64bit (20H2) on a surface laptop 3.

When I try to install it manually via the hardware device, it says that the folder doesn't have driver for the system. Did I miss smth?

Thx for your help and thx for the package, this package doesn't give me errors when I list the remote devices, I tried a different one, which I thought it was the official one and it failed right from the beginning.

addius8, 2020-11-11:
looks to me like they swapped install to install_ude. For me the install worked with`
```
.\usbip.exe install_ude
```
instead of 
```
.\usbip.exe install
```

Chris2011, 2020-11-11:
Oh I overseen it completely. Thx. Now I get this error:

w/o admin rights
```
usbip: error: failed to install vhci driver package: err:5
usbip: error: cannot install vhci driver package
```

w/ admin rights

```
usbip: error: failed to install vhci driver package: err:e0000247
usbip: error: cannot install vhci driver package
```

Maybe a conflict with another driver?

Chris2011, 2020-11-11:
Ok I think I missed the part with the cert. Will do this first. Thought that this is not needed in some case.

Chris2011, 2020-11-11:
Ok yes, completely my fault. I skipped the steps to install the cert. I couldn't find it as I downloaded the assets file as from the other releases. But I see, it is part of the repo. Now after installing the cert and did a bcdedit I could install it via admin rights. Nice thx :).

closed 2020-11-11 by Chris2011

addius8, 2020-11-11:
@Chris2011 do you also have certificate loading errors when binding usb devices in Windows while setting up the windows as server?

Chris2011, 2020-11-11:
https://github.com/cezanne/usbip-win/issues/11#issuecomment-466327436 this comment gave me the answer about the missing step :D.

I'm using a Raspi as the USB Server, not windows. But maybe you also missed a step. I would suggest you also to go step by step through the readme and with admin rights.

addius8, 2020-11-11:
@Chris2011 all my Linux devices are serving completely fine. could you try to bind a device for me. I just want to make sure that its not a repo mistake. Just get the windows devices with
```
.\usbip.exe list -l

   Razer USA, Ltd : unknown product (1532:0114)
 - busid 1-191 (152a:8750)
   Thesycon Systemsoftware & Consulting GmbH : unknown product (152a:8750)
 - busid 1-147 (1532:0032)
   Razer USA, Ltd : unknown product (1532:0032)
 - busid 1-66 (0451:8142)
```

and then try the binding
```
.\usbip.exe bind -b {your BUS-ID}
.\usbip.exe bind -b 1-48
```

Chris2011, 2020-11-11:
So for me, it was just the problem with the missing steps, everything worked fine. And also, I could connect remotely to my usb on my pi. But I have a different setup than you. This is the command, that I used: `usbip.exe attach_ude -r <ip> -b <bus_id>`. The server is running on my pi, I use windows as my client.