cezanne/usbip-win #208: In ude mode, Bulk transfer seem to be slower than Interrupt transfer?

windowsair, 2020-11-11:
Thank you for your excellent work in ude mode!

The following are the average time results of transmission under winusb (Bulk) and HID (interrupt). 

> Take the result of the first transmission here, and transmit the same 2 bytes of data.(In fact, the length of HID transmission is longer, but it takes less time)

1.  Bulk

[image]

2.  Interrupt

[image]

I am not sure what caused the performance loss of bulk transmission.  Do you have any ideas, thanks!:heart:

cezanne, 2020-12-26:
@windowsair : There's no big difference at vhci code level. Did bulk and interrupt use the same device? If not, HW factor might be important. If so, we can suspect driver stacks for each case.

windowsair, 2020-12-26:
Hi, @cezanne

Yes, I use the same hardware for transmission. I'm not sure what caused it

cezanne, 2020-12-26:
@windowsair : bulk case seems to work with more driver stacks including a winusb layer. That would be the reason why bulk is slower.

windowsair, 2020-12-26:
@cezanne Umm, this is probably an important reason. 

Is this related to HID? It seems that under HID, the host is more actively requesting data from the device endpoint

closed 2021-02-13 by windowsair