cezanne/usbip-win #211: Add a timeout or keepalive in case of unexpected disconnection of the client

csphoenix1, 2020-12-13:
Hi :)

When my USBIP client (eg. Windows) crash, or endured an unfortunate disconnection, the device binded on the remote USBIP server is still attached.
So when my client has just restarted, it can't attach the device, because the device is already attached.

Is there a possibility to add a feature to avoid it?
Eg. add the timeout or a keepalive between client and host. In that case, if the client not answer the server, the server will detach the device.
Eg. or, let the client the availability to force a detach/attach with the server ?

Thank you :)

cezanne, 2021-01-12:
@csphoenix1 : There are similar issues: #33 #160

Your suggested methods are possible only through the cooperation of a usbip server.

csphoenix1, 2021-01-12:
Thank Cezanne. I will check that :)

cezanne, 2021-01-14:
@csphoenix1 : Please, check [this](https://github.com/cezanne/usbip-win/wiki/Deployment-Tips#handling-abortive-disconnection).

closed 2021-01-26 by cezanne