cezanne/usbip-win #212: vhci_ude and non-simple USB devices

spog, 2020-12-20:
Hi,
while trying out several USB audio interfaces, i made some changes to the 'vhci_ude' driver:
1. to be able to successfuly unload the driver also after some errors.
2. to get to the state where a non-simple device is recognized and set-up (however not yet operational).
3.  to fix some errors produced by the 'verifier' and debug mode usage.

The modifications are available in my fork 'https://github.com/spog/usbip-win/tree/vhci_ude', but please bear in mind that i am not a Windows expert, so i might have made something completely wrong:)

koudis, 2020-12-20:
@spog thanks for that! Can you create PR from yuor repository to the usbip-win ude branch? So we can see the diff :).

spog, 2020-12-20:
@koudis did the PR:)

cezanne, 2020-12-26:
@spog : Your PR is closed. Maybe this is due to that I removed the vhci_ude branch. That branch is rebased on the master and merged. Please rebase your work on the current master.

spog, 2020-12-26:
@cezanne thanks for the info. I'll try to do the rebase onto the current master.

cezanne, 2021-01-11:
@spog : Please share your issue status.

spog, 2021-01-11:
@cezanne: i am closing the issue - there was some progress made at least for me.

closed 2021-01-11 by spog