cezanne/usbip-win #214: Connecting more than 1 device with ude

RailRanger, 2020-12-21:
This seems to work with the non ude verison, but with ude if I try to connect two devices I will get: 
```
 ./usbip.exe attach_ude -r 192.168.1.32 -b 1-1.1
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0xb7
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6
```
I tried to add another usb controller manually with the have disk option, however when I add the ude_inf file, it says it does not contain a compatible software driver for your device. If the folder contains a driver, make sure it is designed to work with Windows for x64-based systems. 

I am not sure why I cannot install it manually, when using ./usbip.exe install_ude works just fine/

cezanne, 2020-12-26:
@RailRanger : There is the same issue (#191) and had been fixed. [The new release 0.3.1](https://github.com/cezanne/usbip-win/releases/tag/v0.3.1) can resolve your issue.

RailRanger, 2020-12-26:
My bad, thanks so much for working on this project.

closed 2020-12-26 by RailRanger