cezanne/usbip-win #215: ERROR: Open VHCI driver

ImLorio, 2020-12-22:
I am using usbip to connect my laser engraver, last week everything worked.
Today when I wanted to reconnect my engraver laser, and I got an error.

Error:
[image]

I go in device manager and I have this:
[image]
[image]

Who can help me please ? (and sorry for my english, i'm french --')

cezanne, 2020-12-25:
@ImLorio : Did you enable test signing? You said it was working last week, so I suspect that test singing was disabled by something.

ImLorio, 2020-12-31:
I don't know, I followed the readme tutorial

cezanne, 2021-01-04:
@ImLorio : You can check if test signing is enabled such as the below image. Type `bcdedit` in a command window. If enabled, a "Test Mode" watermark is displayed at the left bottom corner of a screen. 

[image]

cezanne, 2021-01-04:
@ImLorio : I found a good youtube tutorial. Please refer: https://github.com/cezanne/usbip-win/issues/64#issuecomment-753598810

cezanne, 2021-01-16:
@ImLorio : I'll close this issue. Please reopen if you want.

closed 2021-01-16 by cezanne