cezanne/usbip-win #219: Host-Client version mismatch

kombolasha, 2021-01-11:
Hi, 
First of all, thank you for all your hard work here; it is very much appreciated!

I followed these steps exactly, and I believe I've installed the VHCI driver correctly; it's in my device manager and looks OK, but when I try to bind a device, I get... 
"usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6"

The usbip version on my Raspberry Pi is 2.0.0, but on the Windows 10 client, it's 1.0.0. Would that be why I'm receiving the error? If so, how do I get them to match?
Thanks!

cezanne, 2021-01-11:
@kombolasha : Even though usbip-win version string does not match with "usbip-utils 2.0", it is no problem. That versioned linux binaries are working good with usbip-win. 
Your error( 0x6 ) in the client means that a server closed the connection. And it's natural that the usbipd did not have error messages. Because usbipd handed over to your kernel module in raspberry Pi. There should be some releated errors in kernel log. Please share the logs.

kombolasha, 2021-01-12:
@cezanne Thanks for the prompt response! I’ll try to upload the log as soon as I get past some other issues that have cropped up.

kombolasha, 2021-01-16:
After a bunch of trial and error, multiple BSOD events when trying to uninstall the driver, and eventually a fresh OS install, I got it to work once and never again. I'm a noob and the kernel log was even more complicated than the process itself, so I gave up and I'm now using Virtualhere, which worked great for my application.
I do want to thank you for your willingness to help and keeping usbip alive for those adept enough to figure it out!

closed 2021-01-16 by kombolasha