cezanne/usbip-win #221: build fails VS2019 Windows 7

kdschlosser, 2021-01-27:
When compiling on Windows 7 x64 SPI I get a bunch of compilation errors. These errors are from the use of structures and preprocessor macros that are only available for Windows 8 and above.

* PUSB_NODE_CONNECTION_INFORMATION_EX_V2
* PUSB_HUB_INFORMATION_EX
* PUSB_PORT_CONNECTOR_PROPERTIES
* IOCTL_USB_GET_HUB_INFORMATION_EX
* IOCTL_USB_GET_PORT_CONNECTOR_PROPERTIES

I have not gone digging through the code to see how reliant the driver is on these structures and macros in order to work properly. I would have to do some research to see if there is a replacement that can be used or possibly defining the macros and structures 

<details>
  <summary>Build log (click to expand)</summary>

```
1>------ Build started: Project: usbip_common, Configuration: Release x64 ------
2>------ Build started: Project: libdrv, Configuration: Release x64 ------
2>Building 'libdrv' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
1>dbgcode.c
1>names.c
1>usbip_common.c
1>getopt.c
1>getopt_long.c
1>usbip_forward.c
2>dbgcode.c
1>usbip_pki_cat.c
1>usbip_pki_sign.c
1>usbip_setupdi.c
1>usbip_stub.c
2>dbgcommon.c
2>devconf.c
2>pdu.c
2>strutil.c
2>usb_util.c
1>usbip_util.c
2>usbd_helper.c
1>usbip_windows.c
1>usbip_network.c
2>libdrv.vcxproj -> \usbip-win-master\Release\x64\libdrv.lib
3>------ Build started: Project: usbip_vhci, Configuration: Release x64 ------
4>------ Build started: Project: usbip_stub, Configuration: Release x64 ------
5>------ Build started: Project: usbip_vhci_ude, Configuration: Release x64 ------
5>Building 'usbip_vhci_ude' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
5>Stamping \usbip-win-master\Release\x64\usbip_vhci_ude\usbip_vhci_ude.inf
5>Stamping [Version] section with DriverVer=01/26/2021,0.3.3.0
5>\usbip-win-master\driver\vhci_ude\usbip_vhci_ude.inf(5-5): warning 1324: [Version] section should specify PnpLockdown=1.
4>Building 'usbip_stub' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
4>stub_cspkt.c
4>stub_dbg.c
4>stub_dev.c
4>stub_devconf.c
4>stub_dispatch.c
4>stub_driver.c
1>usbip_common.vcxproj -> \usbip-win-master\Release\x64\usbip_common.lib
6>------ Build started: Project: usbip, Configuration: Release x64 ------
7>------ Build started: Project: usbipd, Configuration: Release x64 ------
3>Building 'usbip_vhci' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
3>Stamping \usbip-win-master\Release\x64\usbip_vhci\usbip_vhci.inf
3>Stamping [Version] section with DriverVer=01/26/2021,0.3.3.0
3>Stamping \usbip-win-master\Release\x64\usbip_vhci\usbip_root.inf
3>Stamping [Version] section with DriverVer=01/26/2021,0.3.3.0
3>\usbip-win-master\driver\vhci\usbip_vhci.inf(6-6): warning 1324: [Version] section should specify PnpLockdown=1.
3>\usbip-win-master\driver\vhci\usbip_root.inf(6-6): warning 1324: [Version] section should specify PnpLockdown=1.
3>vhci.c
3>vhci_dbg.c
3>vhci_dev.c
3>vhci_devconf.c
3>vhci_internal_ioctl.c
3>vhci_ioctl.c
3>vhci_ioctl_usrreq.c
3>vhci_ioctl_vhci.c
[... 182 lines cut ...]
5>        1 file(s) copied.
5>        1 file(s) copied.
5>........................
5>Signability test complete.
5>
5>Errors:
5>None
5>
5>Warnings:
5>None
5>
5>Catalog generation complete.
5>\usbip-win-master\Release\x64\vhci_ude_cat\usbip_vhci_ude.cat
5>Done Adding Additional Store
5>Successfully signed: usbip_vhci_ude.cat
5>
5>        1 file(s) copied.
5>Deleted file - \usbip-win-master\Release\x64\vhci_ude_cat\usbip_vhci_ude.cat
5>Deleted file - \usbip-win-master\Release\x64\vhci_ude_cat\usbip_vhci_ude.inf
5>Deleted file - \usbip-win-master\Release\x64\vhci_ude_cat\usbip_vhci_ude.sys
5>Done building project "usbip_vhci_ude.vcxproj".
========== Build: 6 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

</details>

cezanne, 2021-02-13:
@kdschlosser : The recent tag version which supports is [v0.1.0](https://github.com/cezanne/usbip-win/releases/tag/v0.1.0). All upper versions do not support Win7 or 8 because usbip-win started to use kernel libraries of Win10.

kdschlosser, 2021-02-13:
maybe a change to the README is needed then.

This is stated right in it. 
```
For Windows 7 users, change usbip_stub and usbip_vhci projects' Target OS version to Windows 7.
Right-click on the project > Properties > Driver Settings > Target OS version > Windows 7
```

cezanne, 2021-02-14:
@kdschlosser : 
> maybe a change to the README is needed then.

Thanks. README should be fixed.
However, conditional build for Win7 or 8 would be more preferable even though those platforms does not support a certain functionalities.

kdschlosser, 2021-02-14:
I would have to do some reading into those macro/functions in the windows API to see what they actually do and if there is a way to do the same thing they do but for windows 7/8 then it is just a matter of checking to see what OS is set. If there isn't anything that can be done to mimic those features then the functions that use them would have to return values that the core program would understand as a not available feature.

closed 2026-09-16 by kdschlosser