cezanne/usbip-win #223: usbip: error: import device

isonupandit11, 2021-02-18:
i'm getting this error in wsl2

``` 
usbip: debug: usbip.c:129:[run_command] running command: `attach'
libusbip: debug: vhci_driver.c:269:[usbip_vhci_driver_open] available ports: 16
libusbip: debug: vhci_driver.c:281:[usbip_vhci_driver_open] available controllers: 1
libusbip: debug: vhci_driver.c:128:[refresh_imported_device_list] controller 0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 0 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 1 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 2 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 3 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 4 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 5 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 6 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 7 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 8 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 9 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 10 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 11 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 12 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 13 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 14 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 15 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:104:[parse_status] exit
usbip: debug: usbip_attach.c:100:[import_device] got free port 0
libusbip: debug: vhci_driver.c:367:[usbip_vhci_attach_device2] writing: 0 3 65617 0
libusbip: debug: vhci_driver.c:372:[usbip_vhci_attach_device2] attach attribute path: /sys/devices/platform/vhci_hcd.0/attach
usbip: debug: sysfs_utils.c:24:[write_sysfs_attribute] error writing to attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed
usbip: error: import device
``` 
debug info on windows
``` 
usbipd: info: starting usbipd (usbip-win 0.3.3)
usbipd: debug: usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbipd: info: listening on 0.0.0.0:3240
usbipd: debug: usbipd_accept.c:65:[do_accept] connection from 172.18.137.94:43880
usbipd: debug: usbipd_accept.c:25:[recv_pdu] received request: 0x8005 - list devices
usbipd: debug: usbipd_list.c:97:[send_reply_devlist] exportable devices: 1
usbipd: debug: usbip_common.c:92:[dump_usb_device] path                 = \\?\usb#vid_0408&pid_a061#5&1e3d05f&0&5#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbipd: debug: usbip_common.c:93:[dump_usb_device] busid                = 1-81
usbipd: debug: usbip_common.c:99:[dump_usb_device] Device(C/SC/P)       = unknown class / unknown subclass / unknown protocol (ef/02/01)
usbipd: debug: usbip_common.c:101:[dump_usb_device] bcdDevice            = 0
usbipd: debug: usbip_common.c:106:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (0408:a061)
usbipd: debug: usbip_common.c:108:[dump_usb_device] bNumConfigurations   = 0
usbipd: debug: usbip_common.c:109:[dump_usb_device] bNumInterfaces       = 0
usbipd: debug: usbip_common.c:111:[dump_usb_device] speed                = Unknown Speed
usbipd: debug: usbip_common.c:113:[dump_usb_device] busnum               = 1
usbipd: debug: usbip_common.c:114:[dump_usb_device] devnum               = 51
usbipd: debug: usbipd_accept.c:41:[recv_pdu] request 0x8005: done: err: 0
usbipd: debug: usbipd_accept.c:65:[do_accept] connection from 172.18.137.94:43882
usbipd: debug: usbipd_accept.c:29:[recv_pdu] received request: 0x8003 - attach device
usbipd: debug: usbipd_import.c:18:[forwarder_stub] stub forwarding started
usbipd: debug: usbipd_import.c:112:[recv_request_import] import request busid 1-81: complete
usbipd: debug: usbipd_accept.c:41:[recv_pdu] request 0x8003: done: err: 0
usbipd: debug: usbipd_import.c:28:[forwarder_stub] stub forwarding stopped
```

cezanne, 2021-02-28:
@isonupandit11 : 
> usbip: debug: sysfs_utils.c:24:[write_sysfs_attribute] error writing to attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed

You seemed to try attach command with no root permission. Please try again as a root.

isonupandit11, 2021-03-05:
Still same issue
'''sudo usbip -d attach --remote=172.17.0.1 --busid=1-55                                       
usbip: debug: usbip.c:129:[run_command] running command: `attach'
libusbip: debug: vhci_driver.c:269:[usbip_vhci_driver_open] available ports: 16
libusbip: debug: vhci_driver.c:281:[usbip_vhci_driver_open] available controllers: 1
libusbip: debug: vhci_driver.c:128:[refresh_imported_device_list] controller 0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 0 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 1 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 2 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 3 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 4 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 5 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 6 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 7 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 8 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 9 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 10 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 11 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 12 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 13 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 14 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 15 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:104:[parse_status] exit
usbip: debug: usbip_attach.c:100:[import_device] got free port 0
libusbip: debug: vhci_driver.c:367:[usbip_vhci_attach_device2] writing: 0 3 65591 0
libusbip: debug: vhci_driver.c:372:[usbip_vhci_attach_device2] attach attribute path: /sys/devices/platform/vhci_hcd.0/attach
usbip: debug: sysfs_utils.c:24:[write_sysfs_attribute] error writing to attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed
usbip: error: import device'''

isonupandit11, 2021-03-05:
Still same issue
```
┌──(kali㉿ASPIRE7)-[~]
└─$ sudo usbip list --remote=172.17.0.1
usbip: error: failed to open /usr/share/hwdata//usb.ids
Exportable USB devices
======================
 - 172.17.0.1
       1-55: unknown vendor : unknown product (148f:7601)
           : \\?\usb#vid_148f&pid_7601#1.0#{fb265267-c609-41e6-8eca-a20d92a833e6}
           : (Defined at Interface level) (00/00/00)

┌──(kali㉿ASPIRE7)-[~]
└─$ sudo usbip attach --remote=172.17.0.1 --busid=1-55
usbip: error: import device

┌──(kali㉿ASPIRE7)-[~]
└─$ sudo usbip -d attach --remote=172.17.0.1 --busid=1-55                                       
usbip: debug: usbip.c:129:[run_command] running command: `attach'
libusbip: debug: vhci_driver.c:269:[usbip_vhci_driver_open] available ports: 16
libusbip: debug: vhci_driver.c:281:[usbip_vhci_driver_open] available controllers: 1
libusbip: debug: vhci_driver.c:128:[refresh_imported_device_list] controller 0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 0 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 1 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 2 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 3 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 4 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 5 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 6 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 7 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 8 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 9 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 10 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 11 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 12 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 13 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 14 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 15 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:104:[parse_status] exit
usbip: debug: usbip_attach.c:100:[import_device] got free port 0
libusbip: debug: vhci_driver.c:367:[usbip_vhci_attach_device2] writing: 0 3 65591 0
libusbip: debug: vhci_driver.c:372:[usbip_vhci_attach_device2] attach attribute path: /sys/devices/platform/vhci_hcd.0/attach
usbip: debug: sysfs_utils.c:24:[write_sysfs_attribute] error writing to attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed
usbip: error: import device ```

cezanne, 2021-03-06:
@isonupandit11 : Sorry for my wrong answer. Your error is not related to root permission. When a non-root user try, following error occurs: 
 ```
usbip: debug: sysfs_utils.c:18:[write_sysfs_attribute] error opening attribute /sys/devices/platform/vhci_hcd.0/attach
```

I have no idea about failure of writing to sysfs. Was there any useful dmesg log in WSL2 ?

isonupandit11, 2021-03-07:
dmesg logs 
```
[45416.132086] WSL2: Performing memory compaction.
[45523.059490] ACPI: bus type USB registered
[45523.060177] usbcore: registered new interface driver usbfs
[45523.060184] usbcore: registered new interface driver hub
[45523.060934] usbcore: registered new device driver usb
[45523.117240] usbcore: registered new interface driver cdc_ether
[45523.126722] usbcore: registered new interface driver rndis_host
[45523.148694] usbcore: registered new interface driver usb-storage
[45523.158182] usbcore: registered new interface driver cdc_acm
[45523.158183] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[45523.168227] usbcore: registered new interface driver ftdi_sio
[45523.168231] usbserial: USB Serial support registered for FTDI USB Serial Device
[45523.191029] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[45523.191033] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 1
[45523.191039] vhci_hcd: created sysfs vhci_hcd.0
[45523.192200] hub 1-0:1.0: USB hub found
[45523.192258] hub 1-0:1.0: 8 ports detected
[45523.192854] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[45523.192855] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 2
[45523.193032] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[45523.193597] hub 2-0:1.0: USB hub found
[45523.193657] hub 2-0:1.0: 8 ports detected
[45537.135044] WSL2: Performing memory compaction.
[45579.688342] vhci_hcd: Failed attach request for unsupported USB speed: UNKNOWN
[45587.766644] vhci_hcd: Failed attach request for unsupported USB speed: UNKNOWN
[45658.141346] WSL2: Performing memory compaction.
[45779.144773] WSL2: Performing memory compaction.
[45835.777265] vhci_hcd: Failed attach request for unsupported USB speed: UNKNOWN
```

cezanne, 2022-06-12:
@isonupandit11 : Are you still interested in this issue? 
usbip stub seemed to deliver wrong USB speed.

isonupandit11, 2022-07-04:
I got it working using usbipd-win project by dorssel following official MS docs.

Kogotoro, 2022-07-06:
> @isonupandit11 : Are you still interested in this issue? usbip stub seemed to deliver wrong USB speed.

catch same (VBox 6.1, Win10, USB-flah-storage) got bcdUSB: 0x0210
but get_speed_from_bcdUSB(...)  switch (bcdUSB)  knows only  0x0100, 0x0110, 0x0200, 0x0250, 0x0300, 0x0310.

closed 2022-07-15 by isonupandit11

CodemanVAN, 2023-01-19:
The same error, @isonupandit11 would you like tell me who you solved this problem?
 ```
usbip: debug: usbip.c:129:[run_command] running command: `attach'
libusbip: debug: vhci_driver.c:269:[usbip_vhci_driver_open] available ports: 16
libusbip: debug: vhci_driver.c:281:[usbip_vhci_driver_open] available controllers: 1
libusbip: debug: vhci_driver.c:128:[refresh_imported_device_list] controller 0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 0 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 1 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 2 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 3 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 4 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 5 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 6 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 7 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 8 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 9 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 10 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 11 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 12 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 13 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 14 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 15 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:104:[parse_status] exit
usbip: debug: usbip_attach.c:100:[import_device] got free port 8
libusbip: debug: vhci_driver.c:367:[usbip_vhci_attach_device2] writing: 8 3 196610 5
libusbip: debug: vhci_driver.c:372:[usbip_vhci_attach_device2] attach attribute path: /sys/devices/platform/vhci_hcd.0/attach
usbip: debug: sysfs_utils.c:18:[write_sysfs_attribute] error opening attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed
usbip: error: import device
```

daanlenaerts, 2023-04-13:
I'm having the same problem at the moment. Does anyone have a solution for this?

isonupandit11, 2023-10-25:
> i just used [this](https://github.com/dorssel/usbipd-win) project following [these](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) documents, faced no such issues.