cezanne/usbip-win #224: Multiple connection support

fatihusta, 2021-02-19:
Hi,
Is it possible to use multiple connection for one shared USB device?

Steveo10, 2021-02-24:
Hi,
I have the same question, I have been doing tests and apparently I think that it cannot be done since when another client connects, the shared resource stops working, but I don't know.

someone has already made the multiple connection work?

fatihusta, 2021-02-24:
Yes, someones says this is not technically possible.

But

I found some devices. Specially smartlogic device shows this is possible. This device share via USB or Network.

I think this device dumping or cloning to virtual device and share. Specially sentinel dongle devices.

https://www.smartlogicltd.com/index.html

https://www.digi.com/products/networking/infrastructure-management/usb-connectivity/usb-over-ip/anywhereusb

Also they are thinking this is not legal. So this is not implemented.

Also this project says it's possible if I understand correctly.

https://github.com/sam88651/UsbHasp

Steveo10, 2021-02-24:
Thanks for your suggestions, I'll be checking the links.

cezanne, 2021-08-15:
@fatihusta : I'm closing this issue. Please reopen if required.

closed 2021-08-15 by cezanne