cezanne/usbip-win #225: Invalid Argument over \driver \usbip test.pfx

Steveo10, 2021-02-25:
Hello

I have tried to compile the project and I get the following error :

invalid argument <.. \ driver \ usbip test.pfx> for property <test certificate>

I have read that the key is placed and then it is modified with 2 points but I keep getting the same error

Will I have the certificate damaged or bad?

[image]

cezanne, 2021-02-28:
@Steveo10 : Did you check the relative path? `..\usbip_test.pfx` seems to be a correct path.

Steveo10, 2021-03-01:
hi @cezanne yes

I have checked the path and made the changes but I always get an error in usbip_vhci and usbip_vhci_ude.

I have placed the same path in the usbip_stub and there I have no error code.

I leave you a picture of the new error, any other recommendation that you can help me with?

[image]

cezanne, 2021-03-02:
@Steveo10 : Well, I believe that build has no problem even though you don't care a test certificate path.
If you did not intend to use an alternate path or another certificate, please remove `driver/vhci/usbip_vhci.vcxproj.user` and try to build again. Certificate setting will be reverted to a usbip-win repository version by removing that file.

cezanne, 2021-06-24:
@Steveo10 : I'll close this issue. Please reopen if you need.

closed 2021-06-24 by cezanne