cezanne/usbip-win #226: Bind fails on Windows 10

crazyquark, 2021-03-09:
No matter what device I try to bind(I want to run the server on Windows) I get this error
(when running with --debug):
usbip: debug: usbip_stub.c:176:[apply_stub_fdo] failed to update driver C:\Users\kris_\AppData\Local\Temp\stuA93A.tmp\usbip_stub.inf ; USB\VID_1A86&PID_7523&REV_0254 ; errorcode: e0000242

cezanne, 2021-06-24:
@crazyquark : Am I too late ? You probably didn't install the test certificate.

cezanne, 2021-06-25:
> @crazyquark : Am I too late ? You probably didn't install the test certificate.

As described in README.md, usbip_test.pfx should be installed onto 2 locations: "Trusted Root Certification Authority" and 
"Trusted Publishers". In your case, there's no USBIP test certificate in "Trusted Publishers".

usbip.exe will give more friendly error message for this issue via 58de484f7ec82480105963d70db6676fe0fa340f.

closed 2021-06-25 by cezanne