cezanne/usbip-win #227: invalid devno / cannot open devno / err:-8

brandonros, 2021-03-12:
```
info: using port 3240 ("3240")
info: starting usbipd (usbip-win 0.3.4)
debug: usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
info: listening on 0.0.0.0:3240

debug: usbipd_accept.c:65:[do_accept] connection from redacted:57001
debug: usbipd_accept.c:29:[recv_pdu] received request: 0x8003 - attach device
debug: usbipd_stub.c:56:[get_device_path] traverse_intfdevs failed, returned: 0
debug: usbipd_stub.c:188:[open_stub_dev] invalid devno: 119
debug: usbipd_import.c:44:[export_device] cannot open devno: 119
debug: usbipd_import.c:92:[recv_request_import] failed to export device: 1-119, err:-8
debug: usbipd_accept.c:41:[recv_pdu] request 0x8003: done: err: -1
```

brandonros, 2021-03-12:
It was working for me on a pair of other PCs yesterday (server + client). I am not sure why this one is failing.

brandonros, 2021-03-12:
i tried to trace it back but it feels like a lot of common "should == 0 for success, -1 for failure" are flipped in some cases without comments as to why:

export_device failed because

open_stub_dev failed / returned INVALID_HANDLE_VALUE

get_devpath_from_devno returned NULL because

get_device_path returned NULL because

traverse_intfdevs returned 0 instead of -1 with walker as walker_devpath because

traverse_dev_info calls walker_devpath and

get_intf_detail returned NULL

cezanne, 2021-06-25:
@brandonros : Indeed, -1 return code of `traverse_intfdevs()` seems to be weird. I'll adjust accordingly. 
Anyway, what's the reason for NULL of `get_intf_detail()`? Wasn't there any debug message?

cezanne, 2021-08-15:
@brandonros : This issue will be closed. Please reopen if you want.

closed 2021-08-15 by cezanne