cezanne/usbip-win #229: Who are you, why do you maintain this project, and how are you funded?

R030t1, 2021-03-16:
Hello maintainer, I appreciate your work. Do you think you could add these things to README?

cezanne, 2021-03-17:
@R030t1 : 
> Hello maintainer, I appreciate your work. Do you think you could add these things to README?

The first and second are not adequate to README. 
The fund acknowledgement has been already mentioned at the tail of README.

closed 2021-03-17 by cezanne

R030t1, 2021-03-17:
Apologies for my oversight, I see the notice at the end. It is sufficient.
I asked for these things as they are helpful when evaluating the quality of a project, the motives of its authors, and the likelihood that it will remain maintained. The notice does all of these for now.