cezanne/usbip-win #230: BSOD while detaching a busy device

cezanne, 2021-03-17:
@spog reported a new BSOD in https://github.com/cezanne/usbip-win/pull/222#discussion_r589054288https://github.com/cezanne/usbip-win/pull/222#discussion_r589054288.
The root cause of this problem is that detach(or plugout) cannot safely prevent USB device I/O. There was a race condition between I/O and plugout. I hope that newly uploaded 493b4b401dd62a1035359636d5afee6b7bafae64 in dev branch would resolve this issue.  

Please test and report. Here is [usbip-win-dev.zip](https://github.com/cezanne/usbip-win/files/6156162/usbip-win-dev.zip).

spog, 2021-03-17:
@cezanne: I've recompiled your 'dev' branch and i can confirm that detach right after previous attach does not cause BSOD anymore.

cezanne, 2021-03-17:
@spog : Thanks for your confirmation.

closed 2021-03-17 by cezanne