cezanne/usbip-win #235: usbip: error: vhci driver is not loaded

zengshing, 2021-04-12:
I have a question when I attached a USB device in Windows 10.
[image]

Zerowalker, 2021-04-15:
same issue here, the drivers install fine as far as i can tell.

Zerowalker, 2021-04-16:
Okay it might be the TESTSIGNING, thought i didn't need it as i installed the drivers just fine.
But tried using it on a virtual machine and when i set it to ON it seems to make the driver load.
Kinda odd though, i thought it only mattered for actually installing it?

cezanne, 2021-06-25:
@zengshing : As @Zerowalker pointed out, you might not run `bcdedit /set TESTSIGNING ON`. In that case, such the error message is generated even though `usbip_test.pfx` has been already installed.

closed 2021-06-25 by cezanne