cezanne/usbip-win #238: Using USBIP-STUB to bind USB camera in Windows7 does not work properly.

guanguancan, 2021-05-08:
The USBIP VHCI side is deployed on Windows Server 2016.
The USBIP STUB is deployed on Windows 7 x64 or Ubuntu x64.
The devices that need to be bound with STUB are cameras of different brands.
The software used to test the camera on the USBIP VHCI side is Microsoft AMCAP.
The code versions of USBIP-win VHCI and USBIP-win STUB are both [0.0.7.](https://github.com/cezanne/usbip-win/releases/tag/v0.0.7)
Each computer is connected by a 100M network.

Status:
When USBIP STUB is deployed on Ubuntu, bind any camera, and everything works fine when AMCAP is run on VHCI side.
When USBIP STUB is deployed on Windows 7, the AMCAP does not show a visible screen even if it runs over 10 minutes. From kernel logs, vhci driver and stub driver simply does isochronous transfers repeatedly. 

How do I troubleshoot the problem? I have a little programming background in WDK and can try to change the code and test it. @cezanne

cezanne, 2021-05-12:
@guanguancan : 
> The code versions of USBIP-win VHCI and USBIP-win STUB are both 0.0.7.

Is there any special reason to use the old version? 
Please try with the latest version. [The up-to-date stub](https://github.com/cezanne/usbip-win/tree/dev)  should be working even with win7. But you need to build against a win7 configuration. Please refer README.

WuXian2021, 2021-07-09:
I got a same problem in Windows 10 using the USB camera. 
The source code is the latest, and USBIP stub is deployed on Windows 2019, USBIP client is deployed on Windows 2019 as well. 
both VHCI and VHCI UDE got a fail.
please let me know how to resolve this problem.
I've attached some stub log when I used to VHCI driver.
I wondering the partially sent is 0, but I am not sure it is correct or else.

00006497	20.07365990	usbip_stub: stub_dispatch: IRP_MJ_READ: Enter	
00006498	20.07365990	usbip_stub: dispatch_read: enter	
00006499	20.07367897	usbip_stub: send_stub_res: sent: cmd:USBIP_RET_SUBMIT,seq:1090,out,ep:0,alen:0	
00006500	20.07416534	usbip_stub: stub_dispatch: IRP_MJ_READ: Enter	
00006501	20.07416725	usbip_stub: dispatch_read: enter	
00006502	20.08786201	usbip_stub: do_safe_completion: status = USBD_STATUS_SUCCESS	
00006503	20.08786964	usbip_stub: done_iso_transfer: sres:seq:1091,data_len:393216,status:STATUS_SUCCESS,usbd_status:USBD_STATUS_SUCCESS	
00006504	20.08787346	usbip_stub: send_stub_res: partially sent: 0 < 2048: seq:1091,data_len:2048	
00006505	20.08798599	usbip_stub: stub_dispatch: IRP_MJ_READ: Enter	
00006506	20.08798790	usbip_stub: dispatch_read: enter	
00006507	20.08799171	usbip_stub: send_stub_res: sent: cmd:USBIP_RET_SUBMIT,seq:1091,out,ep:0,alen:0	
00006508	20.08812332	usbip_stub: stub_dispatch: IRP_MJ_READ: Enter	
00006509	20.08812523	usbip_stub: dispatch_read: enter	
00006510	20.10389519	usbip_stub: do_safe_completion: status = USBD_STATUS_SUCCESS	
00006511	20.10390282	usbip_stub: done_iso_transfer: sres:seq:1092,data_len:393216,status:STATUS_SUCCESS,usbd_status:USBD_STATUS_SUCCESS	
00006512	20.10390663	usbip_stub: send_stub_res: partially sent: 0 < 304640: seq:1092,data_len:304640

cezanne, 2021-12-31:
@WuXian2021 : 
> I am wondering the partially sent is 0, but I am not sure it is correct or else.

This is not an error case. That sort of logs had been shown for my working webcam.

cezanne, 2022-06-11:
@WuXian2021 : Close this issue. Please reopen if you need.

closed 2022-06-11 by cezanne