cezanne/usbip-win #239: FlashPro 6 device not binding

rkhat2, 2021-05-09:
I have a [polarfire SoC FPGA](https://www.microsemi.com/product-directory/soc-fpgas/5498-polarfire-soc-fpga). One of the USB ports on it uses [Flashpro 6](https://www.microsemi.com/product-directory/programming-and-debug/4977-flashpro#device-support).

When I try to bind it in usbip-win server, I get:

```
usbip: error: failed to bind device on busid 1-189 
```

The same USB works with a usbip server in Linux and works with VirtualHere in Windows. Other USBs work fine in usbip-win server. Any reason why this one doesn't work?

I would also like to point out that it shows in Linux as /dev/hidrawN where N is a number. Is hidraw not supported?

cezanne, 2021-10-17:
@rkhat2 : 
> The same USB works with a usbip server in Linux and works with VirtualHere in Windows. Other USBs work fine in usbip-win server. Any reason why this one doesn't work?

usbip-win server (aka stub) still has many issues. `usbip.exe --debug bind -b 1-189` will provide more useful information to resolve your issue.

cezanne, 2021-12-30:
@rkhat2 : Any progress? I'll close this issue. Please reopen if you need.

closed 2021-12-30 by cezanne