cezanne/usbip-win #245: Cannot work with usb2.1 multi interface device.

Skipperddd, 2021-06-24:
I added the bcdUSB0x0210 = UdecxUsbHighSpeed,in get_device_speed.But it does not work.here is the log from ude driver.
00000001	vhci_ude	5092	2780	1	1	06\23\2021-20:02:12:203	    create_fileobject:Enter
00000002	vhci_ude	5092	2780	1	2	06\23\2021-20:02:12:203	    create_fileobject:Leave
00000003	vhci_ude	5092	2780	1	3	06\23\2021-20:02:12:203	    io_device_control:Enter: IOCTL_USBIP_VHCI_GET_PORTS_STATUS
00000004	vhci_ude	5092	2780	1	4	06\23\2021-20:02:12:203	    get_ports_status:Enter
00000005	vhci_ude	5092	2780	1	5	06\23\2021-20:02:12:203	    get_ports_status:Leave
00000006	vhci_ude	5092	2780	1	6	06\23\2021-20:02:12:203	    io_device_control:Leave: STATUS_SUCCESS
00000007	vhci_ude	5092	2780	1	7	06\23\2021-20:02:12:203	    io_device_control:Enter: IOCTL_USBIP_VHCI_PLUGIN_HARDWARE
00000008	vhci_ude	5092	2780	1	8	06\23\2021-20:02:12:203	(EE)vusb_plugin:unknown bcdUSB:210
00000009	vhci_ude	5092	2780	1	9	06\23\2021-20:02:12:203	    io_device_control:Leave: STATUS_SUCCESS
00000010	vhci_ude	8684	5876	9	10	06\23\2021-20:02:12:221	    io_read:Enter: len: 48
00000011	vhci_ude	8684	5876	9	11	06\23\2021-20:02:12:221	    read_vusb:Enter
00000012	vhci_ude	8684	5876	9	12	06\23\2021-20:02:12:221	    io_read:Leave: 0x00000103(STATUS_PENDING)
00000013	vhci_ude	4	5612	7	13	06\23\2021-20:02:12:304	    default_ep_add:Enter
00000014	vhci_ude	4	5612	7	14	06\23\2021-20:02:12:304	    add_ep:Enter: ep_addr=0x0
00000015	vhci_ude	4	5612	7	15	06\23\2021-20:02:12:304	    add_ep:Leave
00000016	vhci_ude	4	5612	7	16	06\23\2021-20:02:12:304	    default_ep_add:Leave: STATUS_SUCCESS
00000017	vhci_ude	4	5612	7	17	06\23\2021-20:02:12:304	    ep_configure:Enter: UdecxEndpointsConfigureTypeDeviceInitialize
00000018	vhci_ude	4	5612	7	18	06\23\2021-20:02:12:304	    store_urbr:urbr: [slc,seq:1,1]
00000019	vhci_ude	4	5612	7	19	06\23\2021-20:02:12:304	    submit_urbr:urb requested: [slc,seq:1,1]: 0x00000103(STATUS_PENDING)
00000020	vhci_ude	4	5612	7	20	06\23\2021-20:02:12:304	    ep_configure:trying to SET CONFIGURATION: 1
00000021	vhci_ude	4	5612	7	21	06\23\2021-20:02:12:304	    ep_configure:Leave: 0x00000103(STATUS_PENDING)
00000022	vhci_ude	8684	5876	9	22	06\23\2021-20:02:12:307	    io_read:Enter: len: 48
00000023	vhci_ude	8684	5876	9	23	06\23\2021-20:02:12:307	    read_vusb:Enter
00000024	vhci_ude	8684	5876	9	24	06\23\2021-20:02:12:307	    io_read:Leave: 0x00000103(STATUS_PENDING)
00000025	vhci_ude	8684	5876	9	25	06\23\2021-20:02:12:392	    io_write:Enter: len: 48
00000026	vhci_ude	8684	5876	9	26	06\23\2021-20:02:12:392	    write_vusb:Enter
00000027	vhci_ude	8684	5876	9	27	06\23\2021-20:02:12:392	    fetch_urbr:Enter: [slc,seq:1,1]
00000028	vhci_ude	8684	5876	9	28	06\23\2021-20:02:12:392	    fetch_urbr:Leave: STATUS_SUCCESS
00000029	vhci_ude	8684	5876	9	29	06\23\2021-20:02:12:392	    write_vusb:Leave: STATUS_SUCCESS
00000030	vhci_ude	8684	5876	9	30	06\23\2021-20:02:12:392	    io_write:Leave
00000031	vhci_ude	0	0	6	31	06\23\2021-20:02:12:426	    ep_purge:Enter: ep->addr=0x0
00000032	vhci_ude	0	0	6	32	06\23\2021-20:02:12:426	    ep_purge:Leave
00000033	vhci_ude	0	0	6	33	06\23\2021-20:02:12:426	    ep_start:Enter: ep->addr=0x0
00000034	vhci_ude	0	0	6	34	06\23\2021-20:02:12:426	    ep_start:Leave
00000035	vhci_ude	4	5612	7	35	06\23\2021-20:02:12:440	    ep_purge:Enter: ep->addr=0x0
00000036	vhci_ude	4	5612	7	36	06\23\2021-20:02:12:440	    ep_purge:Leave
00000037	vhci_ude	4	5612	7	37	06\23\2021-20:02:12:440	    ep_configure:Enter: UdecxEndpointsConfigureTypeEndpointsReleasedOnly
00000038	vhci_ude	4	5612	7	38	06\23\2021-20:02:12:440	    ep_configure:Leave: STATUS_SUCCESS
00000039	vhci_ude	4	2576	2	39	06\23\2021-20:02:12:942	    default_ep_add:Enter
00000040	vhci_ude	4	2576	2	40	06\23\2021-20:02:12:942	    add_ep:Enter: ep_addr=0x0
00000041	vhci_ude	4	2576	2	41	06\23\2021-20:02:12:942	    add_ep:Leave
00000042	vhci_ude	4	2576	2	42	06\23\2021-20:02:12:942	    default_ep_add:Leave: STATUS_SUCCESS
00000043	vhci_ude	4	2576	2	43	06\23\2021-20:02:12:942	    ep_configure:Enter: UdecxEndpointsConfigureTypeDeviceInitialize
00000044	vhci_ude	4	2576	2	44	06\23\2021-20:02:12:942	    store_urbr:urbr: [slc,seq:2,1]
00000045	vhci_ude	4	2576	2	45	06\23\2021-20:02:12:942	    submit_urbr:urb requested: [slc,seq:2,1]: 0x00000103(STATUS_PENDING)
00000046	vhci_ude	4	2576	2	46	06\23\2021-20:02:12:942	    ep_configure:trying to SET CONFIGURATION: 1
00000047	vhci_ude	4	2576	2	47	06\23\2021-20:02:12:942	    ep_configure:Leave: 0x00000103(STATUS_PENDING)
00000048	vhci_ude	8684	5876	10	48	06\23\2021-20:02:12:945	    io_read:Enter: len: 48
00000049	vhci_ude	8684	5876	10	49	06\23\2021-20:02:12:945	    read_vusb:Enter
00000050	vhci_ude	8684	5876	10	50	06\23\2021-20:02:12:945	    io_read:Leave: 0x00000103(STATUS_PENDING)
00000051	vhci_ude	8684	5876	9	51	06\23\2021-20:02:13:362	    io_write:Enter: len: 48
00000052	vhci_ude	8684	5876	9	52	06\23\2021-20:02:13:362	    write_vusb:Enter
00000053	vhci_ude	8684	5876	9	53	06\23\2021-20:02:13:362	    fetch_urbr:Enter: [slc,seq:2,1]
00000054	vhci_ude	8684	5876	9	54	06\23\2021-20:02:13:362	    fetch_urbr:Leave: STATUS_SUCCESS
00000055	vhci_ude	8684	5876	9	55	06\23\2021-20:02:13:362	    write_vusb:Leave: STATUS_SUCCESS
00000056	vhci_ude	8684	5876	9	56	06\23\2021-20:02:13:362	    io_write:Leave
00000057	vhci_ude	0	0	4	57	06\23\2021-20:02:13:463	    ep_purge:Enter: ep->addr=0x0
00000058	vhci_ude	0	0	4	58	06\23\2021-20:02:13:463	    ep_purge:Leave
00000059	vhci_ude	0	0	4	59	06\23\2021-20:02:13:463	    ep_start:Enter: ep->addr=0x0
00000060	vhci_ude	0	0	4	60	06\23\2021-20:02:13:463	    ep_start:Leave
00000061	vhci_ude	4	2576	2	61	06\23\2021-20:02:13:476	    ep_purge:Enter: ep->addr=0x0
00000062	vhci_ude	4	2576	2	62	06\23\2021-20:02:13:476	    ep_purge:Leave
00000063	vhci_ude	4	2576	2	63	06\23\2021-20:02:13:476	    ep_configure:Enter: UdecxEndpointsConfigureTypeEndpointsReleasedOnly
00000064	vhci_ude	4	2576	2	64	06\23\2021-20:02:13:476	    ep_configure:Leave: STATUS_SUCCESS
00000065	vhci_ude	4	6000	11	65	06\23\2021-20:02:13:978	    default_ep_add:Enter
00000066	vhci_ude	4	6000	11	66	06\23\2021-20:02:13:978	    add_ep:Enter: ep_addr=0x0
00000067	vhci_ude	4	6000	11	67	06\23\2021-20:02:13:978	    add_ep:Leave
00000068	vhci_ude	4	6000	11	68	06\23\2021-20:02:13:978	    default_ep_add:Leave: STATUS_SUCCESS
00000069	vhci_ude	4	6000	11	69	06\23\2021-20:02:13:978	    ep_configure:Enter: UdecxEndpointsConfigureTypeDeviceInitialize
[... 36 lines cut ...]
00000106	vhci_ude	8684	5876	10	106	06\23\2021-20:02:15:108	    fetch_urbr:Leave: STATUS_SUCCESS
00000107	vhci_ude	8684	5876	10	107	06\23\2021-20:02:15:108	    write_vusb:Leave: STATUS_SUCCESS
00000108	vhci_ude	8684	5876	10	108	06\23\2021-20:02:15:108	    io_write:Leave
00000109	vhci_ude	0	0	3	109	06\23\2021-20:02:15:210	    ep_purge:Enter: ep->addr=0x0
00000110	vhci_ude	0	0	3	110	06\23\2021-20:02:15:210	    ep_purge:Leave
00000111	vhci_ude	0	0	3	111	06\23\2021-20:02:15:210	    ep_start:Enter: ep->addr=0x0
00000112	vhci_ude	0	0	3	112	06\23\2021-20:02:15:210	    ep_start:Leave
00000113	vhci_ude	4	5612	7	113	06\23\2021-20:02:15:222	    ep_purge:Enter: ep->addr=0x0
00000114	vhci_ude	4	5612	7	114	06\23\2021-20:02:15:222	    ep_purge:Leave
00000115	vhci_ude	4	5612	7	115	06\23\2021-20:02:15:222	    ep_configure:Enter: UdecxEndpointsConfigureTypeEndpointsReleasedOnly
00000116	vhci_ude	4	5612	7	116	06\23\2021-20:02:15:222	    ep_configure:Leave: STATUS_SUCCESS
00000117	vhci_ude	6844	7944	9	117	06\23\2021-20:26:33:886	    create_fileobject:Enter
00000118	vhci_ude	6844	7944	9	118	06\23\2021-20:26:33:886	    create_fileobject:Leave
00000119	vhci_ude	6844	7944	9	119	06\23\2021-20:26:33:886	    io_device_control:Enter: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
00000120	vhci_ude	6844	7944	9	120	06\23\2021-20:26:33:886	    plugout_vusb:plugging out device: port: 0
00000121	vhci_ude	6844	7944	9	121	06\23\2021-20:26:33:886	    vusb_plugout:abort read request
00000122	vhci_ude	6844	7944	9	122	06\23\2021-20:26:33:886	    plugout_vusb:completed to plug out: port: 0
00000123	vhci_ude	6844	7944	9	123	06\23\2021-20:26:33:886	    io_device_control:Leave: STATUS_SUCCESS
00000124	vhci_ude	6844	7944	9	124	06\23\2021-20:26:33:886	    cleanup_fileobject:Enter
00000125	vhci_ude	6844	7944	9	125	06\23\2021-20:26:33:886	    cleanup_fileobject:Leave
00000126	vhci_ude	8684	5876	11	126	06\23\2021-20:26:33:886	    io_read:Enter: len: 48
00000127	vhci_ude	8684	5876	11	127	06\23\2021-20:26:33:886	    io_read:vusb disconnected: port: 0
00000128	vhci_ude	8684	5876	11	128	06\23\2021-20:26:33:886	    io_read:Leave: 0xc000009d(STATUS_DEVICE_NOT_CONNECTED)
00000129	vhci_ude	8684	5876	9	129	06\23\2021-20:26:33:887	    cleanup_fileobject:Enter
00000130	vhci_ude	8684	5876	9	130	06\23\2021-20:26:33:887	    cleanup_fileobject:Leave

Skipperddd, 2021-06-24:
And here is the log from debug_pdu.log:
DUMP: CMD_SUBMIT,seq:1,devid:10002,dir:out,ep:0
  flags:1,len:0,sf:0,#p:0,intv:0
  setup: 0009000000000000
DUMP DONE-------
DUMP: RET_SUBMIT,seq:1,devid:10002,dir:out,ep:0
  st:0,al:0,sf:0,#p:0,ec:0
DUMP DONE-------
DUMP: CMD_SUBMIT,seq:2,devid:10002,dir:out,ep:0
  flags:1,len:0,sf:0,#p:0,intv:0
  setup: 0009000000000000
DUMP DONE-------
DUMP: RET_SUBMIT,seq:2,devid:10002,dir:out,ep:0
  st:0,al:0,sf:0,#p:0,ec:0
DUMP DONE-------
DUMP: CMD_SUBMIT,seq:3,devid:10002,dir:out,ep:0
  flags:1,len:0,sf:0,#p:0,intv:0
  setup: 0009000000000000
DUMP DONE-------
DUMP: RET_SUBMIT,seq:3,devid:10002,dir:out,ep:0
  st:0,al:0,sf:0,#p:0,ec:0
DUMP DONE-------
DUMP: CMD_SUBMIT,seq:4,devid:10002,dir:out,ep:0
  flags:1,len:0,sf:0,#p:0,intv:0
  setup: 0009000000000000
DUMP DONE-------
DUMP: RET_SUBMIT,seq:4,devid:10002,dir:out,ep:0
  st:0,al:0,sf:0,#p:0,ec:0
DUMP DONE-------

cezanne, 2022-06-12:
@Skipperddd : I'm willing to resolve your issue if USB packet logs is available. (USB packet capture tool: https://desowin.org/usbpcap/)

I'm going to close this issue. Please reopen if you need.

closed 2022-06-12 by cezanne