cezanne/usbip-win #246: Debug Assertion Failed when listing USB Devices

KylerFarrar, 2021-06-27:
Using version 0.3.5(dev) on Windows 10 Build 21390 (Insiders), the following error appears:

[image]

Clicking Retry will sometimes get the system to print USB devices, sometimes it just falls through with no output.

KylerFarrar, 2021-06-27:
Small update, clicking ignore will allow it to output devices. Retry does not allow it to continue.

Ithilien, 2021-07-22:
Im not sure if this is a problem with the release or just our download of it; but its a result of corruption [userspace/usb.ids](https://github.com/cezanne/usbip-win/blob/master/userspace/usb.ids); if you replace the contents of that file on your machine with that in the repo, it should resolve the issue. I had the same problem and after replacing my local usb.ids with the one actually in the repo; it didnt have this assertion failure any more.

cezanne, 2021-11-17:
@KKonaOG : I had uploaded the corrupted usb.ids, which was replaced with a fixed one.

closed 2021-11-17 by cezanne