cezanne/usbip-win #247: libusbip and usbip binding error

KylerFarrar, 2021-06-27:
When attempting to attach to a USB Joystick (Saitek ST290 Pro), I receive the following errors:

[image]

After a quick google, I found that it might be due to a version mismatch between WSLs USBIP version and USBIP-WIN version.

As you can see VHCI hcd was enabled as a built-in item inside the WSL2 kernel:
[image]

VHCI hcd successfully went through modprobe in this start_usb.sh script I have to verify connection:
[image]

This error is 1 to 1 with issue #206. I did not know if it was "kosher" to reopen the issue in my name.

teharris1, 2021-07-13:
I have the exact same issue. I have loaded the vhci driver (modprob vhci-hcd) but usbip on WLS2 still throws an error:

```
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
```

psi-cmd, 2021-11-16:
Hello, I think I finally got the inspiration from the answer: <https://askubuntu.com/questions/1234108/unknown-symbol-in-module-or-unknown-parameter-unknown-symbol-usb-hcd-amd-remo> 

It points out that WSL won't change the kernel to self-built version unless you specify the kernel file in .wslconfig under your windows home directory. You should verify kernel built date with `uname -v` and you'll understand. 

I compiled the kernel from wsl2, and copy the `make install` generated kernel from /boot to windows home dir, then specify kernel with documentation from <https://docs.microsoft.com/en-us/windows/wsl/wsl-config#options-for-wslconfig>. and everything works fine.

cezanne, 2022-06-12:
I'm going to close this issue. Thanks to @psi-cmd . Your comment would be very helpful to similar issues.

closed 2022-06-12 by cezanne