cezanne/usbip-win #248: Error: vhci driver is not loaded

vadimgrn, 2021-07-05:
Windows 10 Home 21H1, OS build 19043.1081

"usbip install -w" installs the driver successfully. But if try to attach to some device, you will get "usbip: error: vhci driver is not loaded".

adrwstn, 2021-07-06:
are you running on a VM or through bootcamp by chance, I'm trying to troubleshoot the same issue;
I'm on a MBP running bootcamp

vadimgrn, 2021-07-06:
I'm running Win10 on VirtualBox for Linux.

adrwstn, 2021-07-06:
I wonder if it's just a coincidence were both getting the issue in a VM running on a non-windows native machine i.e. the host OS isn't Windows? If I were to guess, most people testing this are probably running a Windows VM inside of a Windows host OS which might be the reason why this problem hasn't been reported yet (from what I can tell by reading closed issues)
 
I'm Running W10 through Parallels (similar to VirtualBox but for Mac) I can enable test signing in the VM but when I'm running W10 through bootcamp (separate from Parallels) I can't enable test signing so technically I'm not testing this on a 100% native version of W10, or atleast that's how I understand it. From my research i'm thinking this might be the case because Apple uses a compatibility layer to interface with Windows BIOS, prevents the ability to toggle a couple BIOS options that would otherwise be assessable such as disabling secure boot for W10

vadimgrn, 2021-07-06:
I'm sure this is the issue of the driver. I've worked a lot with VBox and can't complain about it.

adrwstn, 2021-07-07:
Have you tried the _ude version? I'm testing the use version out because I experienced OS instability (& blue screen) twice in a row with _wdm version. Not that it should matter but might be helpful for the dev to know that both usbip.exe install -u and usbip.exe install -w result in 'usbip error: vhci driver not loaded' when trying to attach to a device

vadimgrn, 2021-07-07:
vhci_ude driver works for usb 2.0/3.0 flash drives. It works for usb 2.0 webcam if plug in it in 3.0 port, but doesn't work if cam is plugged in 2.0 port. For that reason I'd like to try vhci wdm driver.

vadimgrn, 2021-07-10:
I've fixed this bug in my repo.

cezanne, 2021-07-14:
@vadimgrn :
> vhci_ude driver works for usb 2.0/3.0 flash drives. It works for usb 2.0 webcam if plug in it in 3.0 port, but doesn't work if cam is plugged in 2.0 port. For that reason I'd like to try vhci wdm driver.

Thanks for good information. And I'll check your PR.

vadimgrn, 2021-07-22:
I built the latest master branch, "usbip install -w" doesn't work. You removed something required from my PR.

cezanne, 2021-07-23:
@vadimgrn : I had tested both ude and wdm vhci. Could you describe the problem more?

vadimgrn, 2021-07-23:
if (!has_certificate("USBIP Test")) fails because load_cert_context() calls CertFindCertificateInStore that returns NULL.
**But it works without this check.** I don't know how to install the test certificate correctly, your description "Install driver/usbip_test.pfx (password: usbip)" is not clear (how?). My system didn't ask me where to install it, etc.

cezanne, 2021-07-23:
@vadimgrn : usbip-win assumes that all drivers are built with the `usbip_test.pfx` test certificate. If you build with your own valid certificate, checking `usbip_test.pfx` existence is useless. But currently, usbip.exe requires `usbip_test.pfx` to be installed. Or you should tweak usbip.exe.

`usbip_test.pfx` can be installed by doubling clicking in explorer or using `certutil` command. And please note that the test certificate should be two cert store locations as described in README.

vadimgrn, 2021-07-23:
If the certificate is not installed, there will be the same error with or without this check. But in my case this check is incorrect because it works without it.

cezanne, 2021-07-26:
@vadimgrn : Did you build with your own certificate? I understand that the installation procedure would be improved for that case.

vadimgrn, 2021-07-26:
I use the certificate of usbip. I reinstalled it correctly, now it works. Thank you.

cezanne, 2022-09-07:
@vadimgrn : Close this issue.

closed 2022-09-07 by cezanne