cezanne/usbip-win #250: Error: STATUS_INVALID_PARAMETER when connecting the webcam

myjimmy, 2021-07-11:
Hi, @cezanne 
I can't connect the webcam with this project.
My environment is as follows:
- WebCam: USB 2.0 PC CAMERA
[image]
 
- USB/IP Server: Windows 10 64bit (Version 2004, OS Build 19041.928)
- USB/IP Client: Windows Server 2019 (Version 1809, OS Build 17763.1757)
NOTE: Both Windows 10 and Windows Server 2019 are running on VMware® Workstation 16 Pro.

I installed the vhci-ude driver on Windows Server 2019.
When connecting the webcam from Windows 10 to Windows Server 2019, I got the following error:
- USB/IP Server
[image]

- USB/IP Client
[image]

I attached a zip file of the log files
I compressed the following log files to a zip file and attached it:
- debug_pdu.log: The log file of the usb packet forwarder in USB/IP server
- usbipd.log: The log file of usbipd.exe in USB/IP server
- ude-driver.log: The log file of the vhci-ude driver captured with TraceView in USB/IP client

Please help.
Any suggestion and advice will be helpful.

Regards.

myjimmy, 2021-07-12:
I used the real computer instead of the virtual machine (Windows Server 2019).
Windows 10 64bit (Version 2004, OS Build 19042.985) is running on that computer.
I installed the vhci-ude driver and repeated the above test case.
But I got the same error.
I attached the log files. 
[local-webcam.zip](https://github.com/cezanne/usbip-win/files/6798582/local-webcam.zip)

cezanne, 2021-07-14:
@myjimmy : Your both vhci logs said in common that remote usbip-win servers did not respond. Could you share a usbip-win stub log?

myjimmy, 2021-07-14:
@cezanne Thanks for your reply.
Could you let me know how to capture the log of the usbip-win stub driver?

cezanne, 2021-07-14:
@myjimmy : 

> Could you let me know how to capture the log of the usbip-win stub driver?

Same as [vhci(wdm)](https://github.com/cezanne/usbip-win#how-to-get-windows-kernel-log-for-vhciwdm).

myjimmy, 2021-07-14:
I attached the stub log.
[webcam-stub-log-usb3.1.zip](https://github.com/cezanne/usbip-win/files/6816480/webcam-stub-log-usb3.1.zip)

myjimmy, 2021-07-14:
@cezanne 
And could you let me know how to capture a specific USB device with USBPcap?
Please refer to [this link](https://github.com/desowin/usbpcap/issues/64#issuecomment-879922377).

cezanne, 2021-07-16:
> @cezanne
> And could you let me know how to capture a specific USB device with USBPcap?
> Please refer to [this link](https://github.com/desowin/usbpcap/issues/64#issuecomment-879922377).

I had no idea that USBPcap could capture a specific device. In my case, I used to exploit wireshark filter to view specific device logs from all captures.

cezanne, 2021-07-16:
@myjimmy :
> I attached the stub log.
webcam-stub-log-usb3.1.zip

Thanks for log. But the stub log does not help. It just says that several USB interrupt transfers were cancelled by the vhci. 
What's your device? My webcam(Logitech webcam c170, USB2.0) is working good with vhci(ude) and stub.

myjimmy, 2021-07-16:
My webcam is as follows:

- Device Name: USB 2.0 PC Camera
- Vendor ID: 1908
- Product ID: 2311
- Inf Name: usbvideo.inf
- Device stack:
  - \Driver\ksthunk
  - \Driver\usbvideo
  - \Driver\usbccgp

My webcam isn't a famous brand.

cezanne, 2021-07-23:
@myjimmy : I had compared your vhci log with [mine](https://github.com/cezanne/usbip-win/files/6866315/audio_linux.txt). Your log has several warnings and repetitive unlink packets, which don't appear in mine. I suspect that there was an additional driver installed in your environment. Could you share your driver stack information of a webcam device ? And please retry with the recent [v0.3.5](https://github.com/cezanne/usbip-win/releases/tag/v0.3.5).

myjimmy, 2021-07-23:
@cezanne Thanks for your reply.
How can I get the driver stack information of my webcam device?

cezanne, 2021-07-23:
> How can I get the driver stack information of my webcam device?

You can browse it in device manager. See below attached image.

> And I have the same problem with the recent v0.3.5.

Do you have the same problem with other USB devices? 

[image]

myjimmy, 2021-07-23:
The driver stack information of my webcam device is as follows:
[image]

When using the Arduino device, I can't see this problem.

cezanne, 2021-07-26:
@myjimmy : Can you see the same device stack for the USB composite device(not camera device) which is virtually created by usbip-win? Please refer  my screen.  [image]

WuXian2021, 2021-07-27:
I met the same error with the above.
could you possibly check this issue?
[image]

WuXian2021, 2021-07-27:
I got a some errors as following:
00001501	355.73892212	usbip_stub: control_transfer: seq:175, csp:in,BMREQUEST_CLASS,BMREQUEST_TO_INTERFACE,Unknown request code: 86,wIndex:512,wLength:1,wValue:1792

that's why I updated some codes such as.
[image]
Then I can take the camera in Device manager, but I can't get the camera list on Chrome or Zoom.
[image]
I attached the stub side's log.

[dbgview.log](https://github.com/cezanne/usbip-win/files/6883272/dbgview.log)

could you possibly check this issue?
Thank you in advance.

cezanne, 2021-07-27:
@WuXian2021 : Do you mean that your device was detected after your patch had been applied to a stub ? If so, what's the code for `validate_cspkt_request`?

WuXian2021, 2021-07-27:
Yeah. right. My code is working to avoid the out action to the device when the stub got an unknown class command.

WuXian2021, 2021-07-28:
I captured the URB and got messages such as:
[image]
Thank you.

WuXian2021, 2021-07-28:
And stub side's capture log and device stack are like as:

[image]
[image]

There are no errors on the stub side, it seems.
Could you possibly check the issue?
Thank you.

cezanne, 2021-07-28:
@WuXian2021 : Please share your code for `validate_cspkt_request()`.
And can you test with a linux usbip server? It can help to understand that a usbip-win stub server is a problem or not.

WuXian2021, 2021-07-29:
Both BMREQUEST_CLASS and BMREQUEST_VENDOR are processed by using process_class_vendor_request.
In my case 
reqType is: BMREQUEST_CLASS
reqCode is: 0x86h

I don't know this command is what, and the device is entered the un-succeeded status after run this command, so I added this code to avoid the device panic. After using this code, the device is added to the device manager list, but can't see on the camera list.

static void
process_control_transfer(usbip_stub_dev_t *devstub, struct usbip_header *hdr)
{
	usb_cspkt_t	*csp;
	UCHAR		reqType;

	csp = (usb_cspkt_t *)hdr->u.cmd_submit.setup;

	DBGI(DBG_READWRITE, "control_transfer: seq:%u, csp:%s\n", hdr->base.seqnum, dbg_ctlsetup_packet(csp));

	reqType = CSPKT_REQUEST_TYPE(csp);

	if (BMREQUEST_CLASS == reqType && validate_cspkt_request(CSPKT_REQUEST(csp)) != 0)
	{
		DBGI(DBG_READWRITE, "validate_cspkt_request: seq:%u, csp:%s\n", hdr->base.seqnum, dbg_ctlsetup_packet(csp));
		return;
	}

	switch (reqType) {
	case BMREQUEST_STANDARD:
		process_standard_request(devstub, hdr->base.seqnum, csp);
		break;
	case BMREQUEST_CLASS:
		process_class_vendor_request(devstub, csp, hdr, FALSE);
		break;
	case BMREQUEST_VENDOR:
		process_class_vendor_request(devstub, csp, hdr, TRUE);
		break;
	default:
		DBGE(DBG_READWRITE, "invalid request type:", dbg_cspkt_reqtype(reqType));
		break;
	}
}
int
validate_cspkt_request(UCHAR req)
{
	return dbg_namecode_validate(namecodes_cspkt_request, req);
}

int
dbg_namecode_validate(namecode_t* namecodes, unsigned int code)
{
	int retVal = -1;

	/* assume: duplicated codes may exist */
	for (int i = 0; namecodes[i].name; i++) {
		if (code == namecodes[i].code) {
			retVal = 0;
			break;
		}
	}

	return retVal;
}

static namecode_t	namecodes_cspkt_request[] = {
	K_V(USB_REQUEST_GET_STATUS)
	K_V(USB_REQUEST_CLEAR_FEATURE)
	K_V(USB_REQUEST_SET_FEATURE)
	K_V(USB_REQUEST_SET_ADDRESS)
	K_V(USB_REQUEST_GET_DESCRIPTOR)
	K_V(USB_REQUEST_SET_DESCRIPTOR)
	K_V(USB_REQUEST_GET_CONFIGURATION)
	K_V(USB_REQUEST_SET_CONFIGURATION)
	K_V(USB_REQUEST_GET_INTERFACE)
	K_V(USB_REQUEST_SET_INTERFACE)
	K_V(USB_REQUEST_SYNC_FRAME)
	{0,0}
};

Thank you.

cezanne, 2021-07-30:
@WuXian2021 : 
> And ths usbip-vstus is supporting the camera device?
https://github.com/cezanne/usbip-vstub

I had tried but postponed.

cezanne, 2021-07-30:
@WuXian2021 : Your patch would prevent a stub from replying usbip command. Maybe your issue is related to the problem of stub STALL. Please refer 397b75dcd2a732167972a17565d439d35ab10e2d.

WuXian2021, 2021-08-02:
yeah. I removed the patch because the request is from the PDO. finally, I got this message when I try to open the camera in VLC.exe and chrome.exe.
[image]
Thank you.

WuXian2021, 2021-08-04:
How are you!
I met a new error, I attached the message screenshot.
The pipe is aborted before the starting of the isotransfer, it seems.
Could you possibly check the message?

[image]

Thank you.

cezanne, 2021-08-09:
@WuXian2021 : Thanks.

Your error case is directly related to stub STALL handling problem as described in the upper comment.
I'll try to make a new stub to resolve your issue.

cezanne, 2021-08-10:
@WuXian2021 : Could you test with the [stub driver(binary)](https://github.com/cezanne/usbip-win/files/6959046/usbip_stub.zip)  which handles stall while getting status ? This is a [patch](https://github.com/cezanne/usbip-win/files/6959051/patch.txt) from 3af795da20bc1e5b80d16bdc650de4893fde29fa.

WuXian2021, 2021-08-11:
Thank you for your supports.
Unfortunately, I got to fail and attached the message.
Both sides(stub and client) USB are configured to USB 2.0.
[image]

Thank you.

cezanne, 2021-11-09:
@WuXian2021 : Sorry for too late response. Is your capture about stub or client?

cezanne, 2022-06-10:
@WuXian2021 @myjimmy : Close this issue. Please reopen if you need.

closed 2022-06-10 by cezanne