cezanne/usbip-win #252: Unknown vendor, unknown product (missing USB device names)

MyIsaak, 2021-07-24:
When running the command `usbip list -l` I receive on my Windows 10 19043 the following result:
```
 - busid 1-191 (8087:0029)
   unknown vendor : unknown product (8087:0029)
 - busid 1-98 (13d3:5406)
   unknown vendor : unknown product (13d3:5406)
 <...>
```

Normally one would expect to see actual USB device names, as described in the README. 

Here is my environment:
- Windows 10 v19043
- Windows Studio 2019
- Windows 10 SDK v18362
- MSVC v142
- Test signatures added to "Trusted Root Certification Authority" and "Trusted Publishers" in "Local Computer"
- Test signing enabled and rebooted

cezanne, 2021-07-27:
@MyIsaak : If you copy a up-to-date `usb.ids`, your list would have no unknown. See #151 . 
Yes, usbip-win has too old `usb.ids`. I promise to update at next release.

MyIsaak, 2021-07-28:
Hi @cezanne thanks for the tip. Unfortunately I think the issue lies somewhere else, because I've some of my devices manually in the list. The first one for example 8087:0029 can be found in the _current_ list.
```
8087  Intel Corp.
         // ...
	0029  AX200 Bluetooth
```
To be sure my build isn't the problem, I've reproduced this issue using the precompiled builds on the release page. Could there be any other reason the software isn't reading the ids properly?

cezanne, 2021-07-28:
@MyIsaak : Maybe, the parsing routine(`userspace/lib/names.c`) for usb.ids has a problem. Can you investigate it?

cezanne, 2021-07-29:
@MyIsaak : I've checked the parsing routine is not a problem. A wrong format `usb.ids` might incur your issue. `usbip.ids` in v0.3.5 had a wrong format by my mistake. I have uploaded the corrected version.

MyIsaak, 2021-08-01:
Ok great. I've tested the build in the release page and now works, therefore I'm closing this issue. Thanks!

closed 2021-08-01 by MyIsaak