cezanne/usbip-win #253: Build on ARM64

andyraf, 2021-07-27:
Trying to build on ARM64 and found what I think is a simple issue. The build fails in the post-build step for gencat. The problem is that the vhci and vhci_ude use this post-build step:

    gencat.bat $(OutDir) 10_$(DDKPlatform) $(SolutionDir)\driver\usbip_test.pfx

On x64 builds the second param resolves to 10_X64, and on ARM builds to 10_ARM64.
On ARM builds this fails inside gencat.bat during inf2cat, which doesn't support a 10_ARM64 parameter. However, inf2cat does support ARM64 using other names (e.g. 10_RS3_ARM64, 10_RS4_ARM64, ... 10_VB_ARM64); and each of these also supports an X64 flavor (e.g. 10_VB_X64).

The fix is to either switch to a param version supported by both X64 and ARM64 architectures, e.g. 10_VB_$(DDKPlatform), or create a slightly custom version for ARM64 builds that uses that supported parameter.

cezanne, 2021-07-29:
@andyraf : ARM platform was not considered at all. Some modifications of vcxproj's might be required in order to support ARM. How about making a PR for your changes?

andyraf, 2021-07-30:
Just opened a pull request. Smoke tested the server portion on my Windows ARM-based system attached to an X64 desktop as a client and it appears to be basically functional.

cezanne, 2021-11-18:
@andyraf : Your contribution had been applied to dev branch. I'll close. Please reopen if you need.

closed 2021-11-18 by cezanne