cezanne/usbip-win #257: Crash in UDE client

andyraf, 2021-07-28:
I'm remoting a bunch of serial ports from a modem device. All the ports come up fine in device manager, but the minute I run a terminal program that enumerates the COM ports I hit a blue screen on the x64 client pc. I haven't debugged it beyond looking at the kernel dump, but has anyone else seen this (sorry, I'm super rusty at kernel debugging). On the bright side, the ARM64 server appears to be working like a champ.

4: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

SYSTEM_SERVICE_EXCEPTION (3b)
An exception happened while executing a system service routine.
Arguments:
Arg1: 00000000c0000005, Exception code that caused the bugcheck
Arg2: fffff806407d28dc, Address of the instruction which caused the bugcheck
Arg3: ffffa300c8e5e290, Address of the context record for the exception that caused the bugcheck
Arg4: 0000000000000000, zero.

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.Sec
    Value: 2

    Key  : Analysis.DebugAnalysisProvider.CPP
    Value: Create: 8007007e on ANDYRAF-STUDIO

    Key  : Analysis.DebugData
    Value: CreateObject

    Key  : Analysis.DebugModel
    Value: CreateObject

    Key  : Analysis.Elapsed.Sec
    Value: 6

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 93

    Key  : Analysis.System
    Value: CreateObject

TAG_NOT_DEFINED_202b:  *** Unknown TAG in analysis list 202b

BUGCHECK_CODE:  3b

BUGCHECK_P1: c0000005

BUGCHECK_P2: fffff806407d28dc

BUGCHECK_P3: ffffa300c8e5e290

BUGCHECK_P4: 0

CONTEXT:  ffffa300c8e5e290 -- (.cxr 0xffffa300c8e5e290)
rax=0000000000000000 rbx=ffffbe8479b43870 rcx=0000000000000028
rdx=0000417b864bc788 rsi=ffffbe846a32bdc0 rdi=ffffa300c8e5ed10
rip=fffff806407d28dc rsp=ffffa300c8e5ec90 rbp=fffff80643b4d798
 r8=0000000000000020  r9=ffffbe846a32bf80 r10=fffff806407d2850
r11=0000000000000000 r12=fffff80643b4b0c8 r13=0000000000000000
r14=0000000000000000 r15=fffff80643b4b0c8
iopl=0         nv up ei pl zr na po nc
cs=0010  ss=0018  ds=002b  es=002b  fs=0053  gs=002b             efl=00050246
Wdf01000!FxIrp::GetMajorFunction+0x7 [inlined in Wdf01000!imp_WdfRequestGetParameters+0x8c]:
fffff806`407d28dc 488b88b8000000  mov     rcx,qword ptr [rax+0B8h] ds:002b:00000000`000000b8=????????????????
Resetting default scope

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

PROCESS_NAME:  attacher.exe

STACK_TEXT:  
ffffa300`c8e5ec90 fffff806`43b7900d : 00000000`80000300 ffffa300`c8e5ed58 fffff806`4086ba00 ffffbe84`79b43870 : Wdf01000!imp_WdfRequestGetParameters+0x8c [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 2453] 
ffffa300`c8e5ecf0 fffff806`43b43bd8 : ffffbe84`6a16efa0 ffffbe84`6a16efa0 ffffbe84`79b43870 fffff806`43b4d798 : udecx!UdecxUrbComplete+0x5d
ffffa300`c8e5ed50 fffff806`43b48697 : 00000000`00000000 ffffbe84`77f57110 ffffbe84`6a16efa0 fffff806`43b4d798 : usbip_vhci_ude!complete_urbr+0x5c [C:\repos\usbip-win\driver\vhci_ude\vhci_urbr.c @ 368] 
ffffa300`c8e5ed80 fffff806`43b4847e : ffffbe84`6a8a5590 00000000`00000030 00000000`00000030 ffffbe84`79ad55c0 : usbip_vhci_ude!write_vusb+0x1a7 [C:\repos\usbip-win\driver\vhci_ude\vhci_write.c @ 60] 
ffffa300`c8e5edd0 fffff806`40846692 : ffffbe84`68956920 0000417b`976a9898 0000417b`8913fb88 ffffbe84`68956760 : usbip_vhci_ude!io_write+0xee [C:\repos\usbip-win\driver\vhci_ude\vhci_write.c @ 85] 
ffffa300`c8e5ee20 fffff806`407f6f7d : 00000000`00000000 ffffbe84`76ec0470 ffffbe84`68956760 00000000`00000030 : Wdf01000!FxIoQueueIoRead::Invoke+0x5a [minkernel\wdf\framework\shared\inc\private\common\FxIoQueueCallbacks.hpp @ 159] 
ffffa300`c8e5ee50 fffff806`407d8960 : ffffbe84`79acc000 fffff806`3d516339 ffffffff`ffffffff ffffa300`00000010 : Wdf01000!FxIoQueue::DispatchRequestToDriver+0x1e14d [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3311] 
ffffa300`c8e5eef0 fffff806`407d5d4e : ffffbe84`68956760 ffffbe84`68956700 00000000`00000000 00000000`000007fa : Wdf01000!FxIoQueue::DispatchEvents+0x520 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3125] 
ffffa300`c8e5efc0 fffff806`407d77d7 : 00000000`00000000 ffffbe84`76ec0470 ffffa300`c8e5f131 ffffbe84`68956760 : Wdf01000!FxIoQueue::QueueRequest+0xae [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 2371] 
ffffa300`c8e5f030 fffff806`407d719d : 00000000`00000000 ffffbe84`77e41b60 ffffbe84`76ec0470 00000000`00000000 : Wdf01000!FxPkgIo::DispatchStep1+0x627 [minkernel\wdf\framework\shared\irphandlers\io\fxpkgio.cpp @ 324] 
ffffa300`c8e5f0f0 fffff806`407da867 : ffffbe84`77e41b60 00000000`20206f49 00000193`399612a0 00000000`00000000 : Wdf01000!FxPkgIo::Dispatch+0x5d [minkernel\wdf\framework\shared\irphandlers\io\fxpkgio.cpp @ 119] 
ffffa300`c8e5f150 fffff806`3d5185b5 : 00000000`00000001 ffffbe84`7972ad70 ffffbe84`6a488d90 ffffbe84`77e41b60 : Wdf01000!FxDevice::DispatchWithLock+0x157 [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1447] 
ffffa300`c8e5f1b0 fffff806`3d8e2748 : ffffbe84`77e41b60 00000000`00000000 00000000`00000000 fffff806`3d8f8b8c : nt!IofCallDriver+0x55
ffffa300`c8e5f1f0 fffff806`3d8f88ef : ffffa701`00000000 ffffa300`c8e5f480 00000039`14f7faf0 ffffa300`c8e5f480 : nt!IopSynchronousServiceTail+0x1a8
ffffa300`c8e5f290 fffff806`3d6085b5 : 00000000`00000000 00000000`00000000 00007fff`f7ad7da0 00007ff7`19091640 : nt!NtWriteFile+0x66f
ffffa300`c8e5f390 00007fff`f9d8cf24 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!KiSystemServiceCopyEnd+0x25
00000039`14f7fa48 00000000`00000000 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : 0x00007fff`f9d8cf24

SYMBOL_NAME:  udecx!UdecxUrbComplete+5d

MODULE_NAME: udecx

IMAGE_NAME:  udecx.sys

STACK_COMMAND:  .cxr 0xffffa300c8e5e290 ; kb

BUCKET_ID_FUNC_OFFSET:  5d

FAILURE_BUCKET_ID:  0x3B_c0000005_udecx!UdecxUrbComplete

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {48dea0a8-db66-efc9-d0cd-b31c704f5c07}

Followup:     MachineOwner
---------

Full stack:
nt!KeBugCheckEx
nt!KiBugCheckDispatch+0x69
nt!KiSystemServiceHandler+0x7c
nt!RtlpExecuteHandlerForException+0xf
nt!RtlDispatchException+0x297
nt!KiDispatchException+0x186
nt!KiExceptionDispatch+0x12c
nt!KiPageFault+0x443 (TrapFrame @ ffffa300`c8e5eb00)
Wdf01000!FxIrp::GetMajorFunction+0x7 (Inline Function @ fffff806`407d28dc) [minkernel\wdf\framework\shared\inc\private\km\FxIrpKm.hpp @ 263] 
Wdf01000!FxRequest::GetParameters+0x22 (Inline Function @ fffff806`407d28dc) [minkernel\wdf\framework\shared\core\fxrequest.cpp @ 1275] 
Wdf01000!imp_WdfRequestGetParameters+0x8c [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 2453] 
udecx!UdecxUrbComplete+0x5d
usbip_vhci_ude!WdfObjectDelete+0xd (Inline Function @ fffff806`43b43bd8) [C:\Program Files (x86)\Windows Kits\10\Include\wdf\kmdf\1.15\wdfobject.h @ 743] 
usbip_vhci_ude!free_urbr+0xd (Inline Function @ fffff806`43b43bd8) [C:\repos\usbip-win\driver\vhci_ude\vhci_urbr.c @ 143] 
usbip_vhci_ude!complete_urbr+0x5c [C:\repos\usbip-win\driver\vhci_ude\vhci_urbr.c @ 368] 
usbip_vhci_ude!write_vusb+0x1a7 [C:\repos\usbip-win\driver\vhci_ude\vhci_write.c @ 60] 
usbip_vhci_ude!io_write+0xee [C:\repos\usbip-win\driver\vhci_ude\vhci_write.c @ 85] 
Wdf01000!FxIoQueueIoRead::Invoke+0x5a [minkernel\wdf\framework\shared\inc\private\common\FxIoQueueCallbacks.hpp @ 159] 
Wdf01000!FxIoQueue::DispatchRequestToDriver+0x1e14d [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3311] 
Wdf01000!FxIoQueue::DispatchEvents+0x520 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3125] 
Wdf01000!FxIoQueue::QueueRequest+0xae [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 2371] 
Wdf01000!FxPkgIo::DispatchStep2+0x5ac (Inline Function @ fffff806`407d77d7) [minkernel\wdf\framework\shared\irphandlers\io\fxpkgio.cpp @ 469] 
Wdf01000!FxPkgIo::DispatchStep1+0x627 [minkernel\wdf\framework\shared\irphandlers\io\fxpkgio.cpp @ 324] 
Wdf01000!FxPkgIo::Dispatch+0x5d [minkernel\wdf\framework\shared\irphandlers\io\fxpkgio.cpp @ 119] 
Wdf01000!DispatchWorker+0x6b (Inline Function @ fffff806`407da867) [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1589] 
Wdf01000!FxDevice::Dispatch+0x89 (Inline Function @ fffff806`407da867) [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1603] 
Wdf01000!FxDevice::DispatchWithLock+0x157 [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1447] 
nt!IofCallDriver+0x55
nt!IopSynchronousServiceTail+0x1a8
nt!NtWriteFile+0x66f
nt!KiSystemServiceCopyEnd+0x25 (TrapFrame @ ffffa300`c8e5f400)
0x00007fff`f9d8cf24

andyraf, 2021-07-28:
The m_Irp associated with the request appears to be null.

cezanne, 2022-06-12:
@andyraf : Close this issue. Please reopen if you want.

closed 2022-06-12 by cezanne