cezanne/usbip-win #258: getopt_long bug

FiYHer, 2021-07-28:
`
	for (;;) {
		int	opt = getopt_long(argc, argv, "wuf", opts, NULL);

		if (opt == -1)
			break;

		switch (opt) {
		case 'w':
			only_wdm = TRUE;
			break;
		case 'u':
			only_ude = TRUE;
			break;
		case 'f':
			force = TRUE;
			break;
		default:
			return FALSE;
		}
	}
`

change wuf to w:u:f

`
int	opt = getopt_long(argc, argv, "w:u:f", opts, NULL);
`

FiYHer, 2021-07-28:
for (;;) {
		opt = getopt_long(argc, argv, "pr:l", opts, NULL);

		if (opt == -1)
			break;

		switch (opt) {
		case 'p':
			parsable = TRUE;
			break;
		case 'r':
			ret = list_exported_devices(optarg);
			goto out;
		case 'l':
			ret = list_devices(parsable);
			goto out;
		default:
			break;
		}
	}

opt = getopt_long(argc, argv, "p:r:l", opts, NULL);

FiYHer, 2021-07-28:
https://github.com/cezanne/usbip-win/blob/master/userspace/src/usbip/usbip_install.c#L584
https://github.com/cezanne/usbip-win/blob/master/userspace/src/usbip/usbip_list.c#L58

cezanne, 2021-07-29:
@FiYHer : I don't think so. The current implementation seems to be correct.  A colon appears only if an additional argument(optarg) is required.

closed 2021-07-29 by cezanne