cezanne/usbip-win #260: usbip Win11 is not supported

we2008, 2021-07-31:
usbip Win11 is not supported

e-t-l, 2022-03-04:
@noelhibbard just to clarify (since I'm new to USBIP and this is relevant to my use case), if you have a gamepad physically connected to the server, and you run the USBIP Windows Client to connect to it, then it will appear as if that gamepad is plugged into a USB port on windows?

noelhibbard, 2022-03-04:
> @noelhibbard just to clarify (since I'm new to USBIP and this is relevant to my use case), if you have a gamepad physically plugged into the server device, and you run the USBIP Windows Client to connect to it, then it will appear (on the client device) as if that gamepad is plugged into a USB port on windows?

You are correct. Obviously this isn't a very good use case, I was simply testing and the client and server machines were right next to each other.

e-t-l, 2022-03-04:
> Obviously this isn't a very good use case
Why do you say that? Not to get too off-topic on this thread, but my goal is exactly that: To have two Win11 machines, one with a USB gamepad connected (Xbox One controller, specifically), and for that to appear as a connected USB device on the other Win11 machine. Does this not work very well? Is there high latency or something?

noelhibbard, 2022-03-07:
> 

I figured it would be hard to play a game while physically sitting in one place while the game is running at another location out of view. I guess there could be use cases for this afterall. I only used a gamepad because it was laying around and easy to test with. I didn't perceive any lag but I wasn't really testing for latency. I also tested a webcam connected at another computer over the internet and even that worked well.

e-t-l, 2022-05-01:
@noelhibbard What gamepad did you use? Yesterday I just tested out my setup... USBIP runs correctly on both Win11 devices, the server device was able to bind the wired Xbox One controller, and the client device was able to detect and attach it. Then, the controller showed up under Device Manager on the client side (with no driver errors, too. It appears to be Connected and functioning normally). It also showed up in Control Panel > Devices. The only problem is, it doesn't seem to work. Button presses on the physical controller do not appear to have any effect on the client device, and if I run the Xbox Accessories app on the client (which normally will display all connected Xbox One controllers), it detects none. Do you have any ideas what I could do differently here?

noelhibbard, 2022-05-02:
@e-t-l , I also tested with an Xbox One controller. My testing only when as far as seeing button presses within the joystick control panel (```C:\WINDOWS\system32\joy.cpl```). I didn't run any actual games.

cezanne, 2022-06-12:
I think that this issue can be closed. The issue of @e-t-l  is not fundamentally related to Win11. 
Great thanks to @noelhibbard for resolving the issue.

closed 2022-06-12 by cezanne