cezanne/usbip-win #261: attacher.exe

kevindastick, 2021-08-09:
Im trying to attach the usb from the sever with Command Prompt running in administrator and everytime i run it it gives me a error saying:

C:\WINDOWS\system32>D:/usbip/usbip.exe attach -r 192.168.1.123 -b 1-210
usbip: error: attacher.exe not found

I have the exe in the same location as usbip, unless i need to have it in a specific location?

cezanne, 2021-08-10:
@kevindastick : Thanks for your reporting. Sadly, current usbip.exe does not check attacher.exe based on usbip.exe path. You should change the directory into `D:\usbip`.
I'll fix the problem.

kevindastick, 2021-08-10:
I dragged the attacher.exe into my windows32 files and it now works, so i guess that will be a temporary fix for now. Thank you!

closed 2021-08-22 by kevindastick

iDustbin, 2022-03-01:
not working for me :/ attacher.exe not found