cezanne/usbip-win #263: Does it require an existing controller/hub ?

lgmorand, 2021-08-23:
Sorry for the dumb question (or the incorrect terminology) but this subject is complex to understand for me :)

I'm trying to get it work on an Azure Windows Virtual Machine. Since there is no USB on the hardware, there is no USB controller listed on the VM.

For instance: > usbip.exe list -l returns nothing

Does the driver emulates everything or does it needs to have an existing usb controller to work and emulate a second one ?

Thanks

cezanne, 2021-08-23:
@lgmorand : Do you intend to make your local USB to work in Azure? If yes, you are required to install a usbip-win vhci driver. But I don't know if it's possible to install a custom driver on Azure.
If you want a USB server in Azure, a stub driver should be installed. That driver can be installed without a usb controller, but it's useless. (There's no usb in Azure as you described)

lgmorand, 2021-08-24:
Thanks @cezanne 

we are full admin of the VM and we got weird message but by trying again we were able to move forward

closed 2021-08-24 by lgmorand