cezanne/usbip-win #264: Repeated device binding

rprokop, 2021-08-24:
There is no check if device is already bound. And repeated binding usually completed successfully:
C:\projects\usbip-win\Debug\x64>usbip.exe -d bind -b 1-112
usbip: debug: usbip.c:151:[run_command] running command: bind
usbip: info: bind device on busid 1-112: complete

C:\projects\usbip-win\Debug\x64>usbip.exe -d bind -b 1-112
usbip: debug: usbip.c:151:[run_command] running command: bind
usbip: info: bind device on busid 1-112: complete

Next unbinding always fails:
C:\projects\usbip-win\Debug\x64>usbip.exe -d unbind -b 1-112
usbip: debug: usbip.c:151:[run_command] running command: unbind
usbip: debug: usbip_stub.c:204:[rollback_driver] failed to rollback driver: 103
usbip: error: failed to unbind device on busid 1-112

0x103  = ERROR_NO_MORE_ITEMS
But, in fact, the device was not unbound 

I found only manual way to unbind device.
1. Run "pnputil -f -d oem#.inf". Where number in "oem#.inf" 
The number can be viewed in the properties of the device.
2. "Remove Device" on context menu in the "Devices and Printers".
3.  Remove and insert the physical  device. 

Original usbip on linux prevents repeated bindings
root@host:/home/user# usbip bind -b 1-1.6
usbip: info: bind device on busid 1-1.6: complete
root@host:/home/user# usbip bind -b 1-1.6
usbip: error: device on busid 1-1.6 is already bound to usbip-host

Is there a way to prevent repeated device binding in usbip-win?

rprokop, 2021-08-31:
I modified userspace/lib/usbip_stub.c to insert the check before stub driver installation:
```c
static int
walker_attach(HDEVINFO dev_info, PSP_DEVINFO_DATA pdev_info_data, devno_t devno, void *ctx)
{
	devno_t	*pdevno = (devno_t *)ctx;

	if (devno == *pdevno)
	{
		if(is_service_usbip_stub(dev_info, pdev_info_data))	{
			err("device on busid 1-%d is already bound to usbip-stub", (unsigned)devno);
			return ERR_STATUS;
		}

		int ret = apply_stub_fdo(dev_info, pdev_info_data);
		if (ret < 0)
			return ret;
		return 1;
	}
	return 0;
}
```

cezanne, 2022-06-12:
@rprokop : Please refer to https://github.com/cezanne/usbip-win/issues/311#issuecomment-1152911098

cezanne, 2022-09-07:
@rprokop : close this issue. Please reopen if you need.

closed 2022-09-07 by cezanne