cezanne/usbip-win #267: unable to detach devices

samzhou, 2021-08-30:
When I detach devices, it shows succesfull. But I can still find it using `usbip port` :

```py
PS D:\bin\usbip-win> usbip -d detach -p 3
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
port 3 is succesfully detached

PS D:\bin\usbip-win> usbip port
Imported USB devices
====================
Port 00: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
Port 01: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
Port 02: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
Port 03: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
```

I can try to restart the windows to fix it at last week. But today I can see imported devices without any operation after rebooting.
```py
PS D:\bin\usbip-win> usbip list -r $env:USBCON_HOST
Intel Corp.
Exportable USB devices
======================
  1-1.3.6.1: unknown vendor : unknown product (163c:548c)
           : (Defined at Interface level) (00/00/00)

PS D:\bin\usbip-win> usbip port
Imported USB devices
====================
Port 00: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
Port 01: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
Port 02: <Device in Use> at Full Speed(12Mbps)
       unknown vendor : unknown product (163c:548c)
       ?-? -> unknown host, remote port and remote busid
           -> remote bus/dev ???/???
```

I don't know how can it detach the device, pls help me. thanks!

Server: Ubuntu Server on Raspberry Pi
Client: Windows 10.0.18382,  usbip-win 0.3.5

samzhou, 2021-08-30:
when i passed different port params, but i got the same device path in debug info(in fact they didn't take affect):

```py
PS D:\bin\usbip-win> usbip -d  detach -p 0
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
port 0 is succesfully detached
PS D:\bin\usbip-win> usbip -d  detach -p 1
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
port 1 is succesfully detached
PS D:\bin\usbip-win> usbip -d  detach -p 2
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
port 2 is succesfully detached
PS D:\bin\usbip-win> usbip -d  detach -p 3
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
port 3 is succesfully detached

PS D:\bin\usbip-win> usbip -d  detach -p -1
usbip: debug: usbip.c:151:[run_command] running command: detach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&0#{d35f7840-6a0c-11d2-b841-00c04fad5171}
all ports are detached
```

samzhou, 2021-08-30:
usbip devices are following: (via `devcon find USB*`)
```py
USB\ROOT_HUB&VID_1209&PID_8250&REV_0000\2&62C1C07&0         : usbip-win VHUB
ROOT\USBIP\ROOT                                             : usbip-win VHCI Root
USBIPWIN\VHCI\1&19724A8B&0                                  : usbip-win VHCI
```

tonyho, 2021-12-16:
Same problem, I use the Ubuntu 18.04.6.

Charlesjean, 2023-01-29:
Same problem here. Do you find any solution about this? @samzhou @tonyho