cezanne/usbip-win #271: error: failed to unbind device, and I don't know how to use the device anymore  :-(

cakira, 2021-09-04:
Hi!
I had an error very similar to this one: https://github.com/cezanne/usbip-win/issues/110
I'm using Windows 10 + WSL 2 and I had a successful experience to export a serial device before. So, this time tried to export an embedded system board to WSL.

O Windows I get:

    PS C:\WINDOWS\system32> & 'C:\Program Files\usb_ip\usbip' list -l
    [...]
    - busid 1-169 (1357:0089)
      P&E Microcomputer Systems : OpenSDA - CDC Serial Port (1357:0089)
    [...]

Then `'C:\Program Files\usb_ip\usbip' bind --debug --busid=1-169`

In WSL, I was able to attach the device, but even so I wasn't able to use it, due to some error in my tool.

So, I gave up and tried to use my board on Windows again, but so far without success. In unbind I had an error and after using the customized version for unbind debug (version listed in the ticket above) I got:

    PS C:\WINDOWS\system32> & 'C:\Program Files\usb_ip\usbip_unbind.exe' unbind --busid=1-169
    usbip: error: failed to rollback driver: 259
    usbip: error: failed to unbind device

And I can't use the device anymore. I tried rebooting my PC and tried reinstalling the driver, so far without success.
Any sugestion?

cakira, 2021-09-09:
Following information from https://github.com/cezanne/usbip-win/issues/264 and https://github.com/cezanne/usbip-win/issues/228, I ran `pnputil -f -d oem35.inf` (I discovered the number opening the Windows Device Manager, right click in the drive, Properties, then looking at the tab Events). In Device Manager, the driver was still listed, even if I disconnected the device and rebooted the PC.
So, I disconnected it with `pnputil /remove-device ROOT\JUNGO\0000 /subtree /reboot` (I couldn't disconnect it the same way as https://github.com/cezanne/usbip-win/issues/264 because my device is not listed as a printer). After checking that the device disappeared in Device Manager, I re-plugged the device. The device is shown again in Device Manager, but it still doesn't work.

cakira, 2021-09-09:
The problem was solved. I can't really precise what I did, but among the steps, I did:

    1. Go into Device Manager in Windows,
    2. In the View menu, select "Show hidden devices"
    3. Find the USB/IP STUB device under System devices,
    4. Uninstalled the device

Then I uninstalled my device's driver a couple times and reinstalled, did some usbip bind and some usbip unbind. In one of these attempts, the device driver started working in Windows as expected.

closed 2021-09-09 by cakira

amanvats88, 2022-06-26:
I was able to get out of this problem in a slightly different but essentially the same way as @cakira did.

For me, the USB/IP Stub device was not listed under System Devices, but under Human Interface Devices. I uninstalled it, and it disappeared, only to be found again under the System Devices. Tried uninstalling the device several times and bind/unbind commands in powershell repeatedly with no success.

Finally, as a last ditch effort I uninstalled the device and along side I checked the "Delete the driver software" option before hitting the uninstall button. Then unplugged the USB dongle and plugged it in back to add the device. It was working now.