cezanne/usbip-win #272: Endpoint 0 stalled if attached device is idle

skajacore, 2021-09-07:
Attaching remotely to a mouse, the device successfully attaches from Win 10 and the driver installs, as long as I keep moving the mouse wheel, it remains connected. If I leave it idle, an error on the host saying the endpoint stalled and the device stops working. 

I've also tried this with a bluetooth dongle (my real application), and as long as it is scanning and trying to connect to nearby devices it remains connected, otherwise when idle, it connects and disconnects repeatedly until the USB Root Hub device throws an error that it could not reset. The USB device appears normally in device manager in the bluetooth group, at the time it is connected

Using a linux PC as usb host device version (usbip-utils 2.0)

cezanne, 2021-09-28:
@skajacore : 
I agree vhci log is not pretty good to understand. However, your log looks a bit more complicated. How did you extract it? 
Here is my log from TraceView.exe.
```
00000036        vhci_ude        4       6340    1       36      07\12\2021-05:58:33:744     store_urbr_urb:[urb,seq:2,URB_FUNCTION_CONTROL_TRANSFER_EX]
00000037        vhci_ude        4       6340    1       37      07\12\2021-05:58:33:744     submit_urbr:urb requested: [urb,seq:2,URB_FUNCTION_CONTROL_TRANSFER_EX]: 0x00000103(STATUS_PENDING)
00000038        vhci_ude        4       6340    1       38      07\12\2021-05:58:33:744     internal_device_control:Leave: 0x00000103(STATUS_PENDING)
00000039        vhci_ude        916     3964    1       39      07\12\2021-05:58:33:745     io_read:Enter: len: 48
00000040        vhci_ude        916     3964    1       40      07\12\2021-05:58:33:745     read_vusb:Enter
00000041        vhci_ude        916     3964    1       41      07\12\2021-05:58:33:745     io_read:Leave: 0x00000103(STATUS_PENDING)
00000042        vhci_ude        916     3964    1       42      07\12\2021-05:58:33:795     io_write:Enter: len: 52
```

Anyway, your first 3 warnings seem to be no problem but last warning logs are mysterious.
The linux server might send wrong packets. Was there any clue in your usbip server ?

newprophet, 2024-06-06:
@skajacore Did you fix the bug?