cezanne/usbip-win #284: usb audio class device

yclin-ee, 2021-11-04:
Hello,
I am try to use usbip-win connecting UAC device on Windows 10 both ends.
The two ends could establish connection successfully. But when playing music, it sounds discontinuous and glitches. 
I even did the testing on Ubuntu 18.04 x86_64(server) and Windows 10(client) then found client side always issue Clear_Feature control packet every 10 isochronous packets in one URB transaction.
`[   58.713439] usbip-host 1-1: usb_clear_halt done: devnum 2 endp 6`

Does anybody encountered this before? Or any idea for fixing this issue?

Sok00, 2021-11-09:
try disable `usb auto suspended` in System / Bios
check the `CPU freq. and loade` when transferring. Change the maximum transmission `delay of packets`. Check the network packet priority and process priority.
[QoS Class Identifier (QCI) Wiki Link](https://en.wikipedia.org/wiki/QoS_Class_Identifier)

myjimmy, 2022-04-26:
@Sok00 Please let me know how to change the maximum transmission `delay of packets`.

Carrot-shreds, 2025-07-25:
I solved a similar issue (between an Ubuntu server and a Windows 11 client) by increasing `usbfs_memory_mb` to 2048:  
`sudo vim /sys/module/usbcore/parameters/usbfs_memory_mb`
Then change the value to 2048.
Reference: [Increase usbfs memory limit in Ubuntu](https://importgeek.wordpress.com/2017/02/26/increase-usbfs-memory-limit-in-ubuntu/)