cezanne/usbip-win #285: Uninstall driver command fails on Windows 11

bardgames, 2021-11-10:
Hello,

I am attempting to follow the [README.md](https://github.com/cezanne/usbip-win#readme) installation instructions and have encountered the following issue:

I have installed the certificate to both CAs, enabled secure signing by turning on Test Mode via bcdedit.exe, and successfully attached 3 USB devices from my Raspberry Pi over WiFi to my Windows 11 Home x64 laptop (Build 22000.co_release.210604-1628)

However, when I run the usbip.exe uninstall command, the prompt just hangs. If I close out and reboot my PC, then try again, it does the same thing.

Additionally, is it possible to eventually to leave test signing permanently disabled and re-enable my secure boot settings once I've successfully attached these devices for the first time? Or do I just need to leave test signing on permanently in Test Mode if I want to use USB over IP? I'm hoping that I can find a compromise between security and convenience.

Sok00, 2021-11-12:
Run PowerShell or CMD as an Administrator to install or uninstall driver.

bardgames, 2021-11-12:
I run all of my commands through the PowerShell admin. command prompt from
within the directory itself.

Could there be another permissions issue in play that I'm missing? I've
moved the folder to C:\Drivers.

e-t-l, 2022-03-04:
> I've moved the folder to C:\Drivers.

Just guessing here, but if it's like most programs, the installation path matters. e.g. you wouldn't move an installation folder from Program Files onto your Desktop and expect the app to still work. 

In other words, have you tried moving the folder back to the exact location it was when you originally installed it, and then try uninstalling? The uninstallation process may be looking for "usbip.exe" in the wrong location...