cezanne/usbip-win #289: attach OK, but no response after several minute

lededev, 2021-11-30:
The device I using is a Lefant F2s 2.4G Wireless Motion Game Mini Keyboard Fly Air Mouse.
[image]

lededev, 2021-11-30:
It may caused by attacher.exe memory leak, after no response, attacher.exe memory extend from 300KB to 2,097,468KB

agarity, 2022-01-20:
I see this same behavior too.  soon as device stops responding attacher.exe skyrockets from a few hunderd kb to 3.6gb usually happens within 120 seconds of attaching device.

vadimgrn, 2022-01-29:
Try this build
https://github.com/vadimgrn/usbip-win/releases/download/v0.7.5.1/usbip-win-0.7.5.1.7z

lededev, 2022-01-30:
@vadimgrn Your new build support xbox 360 game controller no problem, I used 15 minutes still work.
But a mouse and keyboard combination device(measy rc7) does not work after attach.
[image]
Device find, but no mater mouse moving and click, or keyboard buttons can not work.

vadimgrn, 2022-01-30:
You could get vhci driver log as described in Readme.md. Refer to the project's page of my repo for instructions.

agarity, 2022-01-31:
@vadimgrn first round of tests, the build seems to resolve my issue.  Thanks for putting this together!