cezanne/usbip-win #291: "USBIP Test" certificate is not found

joestarwalker, 2021-12-03:
Hello!
I've tried to install usbip 0.3.5 on Windows 10 according to the readme. And received the message in the title after running 'usbip.exe install' with cmd. I set up certificate with 'driver/usbip_test.pfx' previously. 
Could you please help me what can it be and how can I fix it?
Huge thanks.

Aussiedler, 2021-12-16:
Same issue here, tried many times with the same version to install usbip.exe via powershell/cmd as admin... i'm in windows 11. Please help!

Aussiedler, 2021-12-16:
my error message is: 

usbip: error: "USBIP Test" certificate not found. Please install first!

But the installation was succefully and the test mode also starts without problems. After reboot i get the error messeage ...

Aussiedler, 2021-12-20:
ok i have to read carefully... in the readme are these two line which solved my prob... 

Certificate should be installed into
"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)

I done it first like in this video, but here the guy only install the certificate in the default path... in his case it should work, because in the past he did it on the right way and forgot to show this one in the tut... I think...

https://www.youtube.com/watch?v=lsMsl8LEZvs&t=50s

bihailantian655, 2022-05-26:
i have the  same  issue  did  you  solve it

amanvats88, 2022-06-26:
It wasn't working for me too. It did when the certificate was installed in **BOTH** :-
1. "Trusted Root Certification Authority" in "Local Computer" (not current user)
2. "Trusted Publishers" in "Local Computer" (not current user)

paulpv, 2022-09-14:
@joestarwalker think this ticket can be closed?

cezanne, 2022-12-28:
@joestarwalker I'll close this issue. Please reopen if you want.

closed 2022-12-28 by cezanne