cezanne/usbip-win #292: mic device attach not work

xux1217, 2021-12-11:
Hi, 

  thanks for your job.

  now I want to know, is there have any restrict about the mic device?

  I use linux server to bind a mic device, and attach from windows client, the mic is attach successful, but not work. I can see the mic in device manager, and seems the driver is not problem.

  I have tested in the same server and other linux usb/ip client, the mic is work.

  usb/ip win client version: 0.3.5
  usb/ip linux server version: usbip (usbip-utils 2.0)

xux1217, 2021-12-14:
Can someone help test it? I want to know if this problem is my problem or a software bug.

thanks very much !

vadimgrn, 2022-01-29:
Could you please try this build?
https://github.com/vadimgrn/usbip-win/releases/download/v0.7.5.1/usbip-win-0.7.5.1.7z

xux1217, 2022-02-09:
it's not work, I just install the windows client, and maybe the device bind successful(in the linux server log), but the ".\usbip.exe port" is not any device, and detach fail.