cezanne/usbip-win #294: Win10 system random crash while communicating with multiple HID devices on RaspberryPI

luepipo, 2022-01-14:
Client: Win10
Server: RPi

We tried to connect multiple HID devices from RPI to Win10 via USBIP v0.3.1.
We encountered several times win10 system crash (no BSOD found, but system reboot) while we left the devices running themselves.

After capturing the minidump and trying to analyzing this by WinDBG, we got below result.
[randomReset.txt](https://github.com/cezanne/usbip-win/files/7870710/randomReset.txt)

I already checked recent commits but still had no idea about this problem.

Here are from the Windbg dump

`FAULTING_SOURCE_LINE:  D:\usbip-win\driver\vhci\vhci_dev.c

FAULTING_SOURCE_FILE:  D:\usbip-win\driver\vhci\vhci_dev.c

FAULTING_SOURCE_LINE_NUMBER:  35

FAULTING_SOURCE_CODE:  
    31: 	if (status != STATUS_BUFFER_TOO_SMALL) {
    32: 		DBGE(DBG_GENERAL, "failed to get device property size: %s\n", dbg_ntstatus(status));
    33: 		return NULL;
    34: 	}
>   35: 	value = ExAllocatePoolWithTag(PagedPool, buflen, USBIP_VHCI_POOL_TAG);
    36: 	if (value == NULL) {
    37: 		DBGE(DBG_GENERAL, "failed to get device property: out of memory\n");
    38: 		return NULL;
    39: 	}
    40: 	status = IoGetDeviceProperty(pdo, prop, buflen, value, &buflen);
`

Thanks,

vadimgrn, 2022-01-29:
Try this build
https://github.com/vadimgrn/usbip-win/releases/download/v0.7.5.1/usbip-win-0.7.5.1.7z

luepipo, 2022-02-07:
Thanks for your reply,
I tested your build but driver seems not to work normally.
[image: image.png]
Does anything I miss?

Thanks,

Lue

mikevador02, 2022-02-09:
Hello,

Thanks for this solution working fine but client crash randomly
I use it with usb device on another windows 10 64 pc (server pc) on wired gigabit LAN.

I have the same random crash on client windows 10 64  (sometime after 5 minutes sometimes after 30 minutes).

I have compiled the project with VS 2019 x64 on the client computer.

my windbg file is 

AULTING_SOURCE_LINE:  C:\usbip-win-master\driver\vhci\vhci_read.c

FAULTING_SOURCE_FILE:  C:\usbip-win-master\driver\vhci\vhci_read.c

FAULTING_SOURCE_LINE_NUMBER:  809

FAULTING_SOURCE_CODE:  
   805: 	DBGI(DBG_READ, "store_urbr_partial: Enter: urbr: %s\n", dbg_urbr(urbr));
   806: 
   807: 	irpstack = IoGetCurrentIrpStackLocation(urbr->irp);
   808: 	urb = irpstack->Parameters.Others.Argument1;
>  809: 	code_func = urb->UrbHeader.Function;
   810: 
   811: 	switch (code_func) {
   812: 	case URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER:
   813: 		status = store_urb_bulk_partial(urbr->vpdo, irp, urb);
   814: 		break;

SYMBOL_NAME:  usbip_vhci!store_urbr_partial+58

MODULE_NAME: usbip_vhci

IMAGE_NAME:  usbip_vhci.sys

Thanks in advance for help.

vadimgrn, 2022-03-14:
Try this, please https://github.com/vadimgrn/usbip-win2/releases