cezanne/usbip-win #296: vhci driver is not loaded for signed drivers

knavels, 2022-02-03:
We've asked our client to sign the driver and we installed the driver properly but I still get `vhci driver is not loaded` error when I want to attach a new device, any experience about this?

vadimgrn, 2022-02-03:
Try this build
https://github.com/vadimgrn/usbip-win/releases

knavels, 2022-02-04:
@vadimgrn I'm glad that I know there is a repo with active development for windows client part, I tested your build unfortunately it didn't help, I checked the Driver though it is signed and in the Driver tab (Device manager, Properties submenu from context menu for the installed driver), it shows the Signature data, but in the General tab it says that: `Windows cannot verify the digital signature for the drivers required for this device. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Code 52)`

vadimgrn, 2022-02-04:
@knavels [knavels](https://github.com/knavels) Thank you for your feedback! Indeed I haven't change code that installs drivers, but vhci driver itself is improved significantly.

cezanne, 2022-09-07:
@knavels : close this issue. Please reopen if you need.

closed 2022-09-07 by cezanne