cezanne/usbip-win #298: attacher.exe not found :/

iDustbin, 2022-03-01:
usbip.exe attach -r 10.0.0.22 -b 2-2 
attacher.exe not found :/ 
can u please help here

stevogray, 2022-03-09:
Attacher.exe needs to be in the working directory when you run usbip.exe.
Try changing the current working directory to the directory where you have usbip.exe and attacher.exe before running the command

e-t-l, 2022-06-17:
@cezanne @iDustbin This issue can probably be closed. The above suggestion solves the issue. Find the folder that attacher.exe is kept in (should also have usbip.exe and all the other files), and execute `cd C:\path\to\folder\` to change your CLI's working directory.

cezanne, 2022-06-18:
@e-t-l @iDustbin : Current working directory would not be desirable as a path of attacher.exe.  92cd6c148a03053e30430bf593b56303b3e889f0 changes attacher search mechanism.

This issue is well deserved to be closed.

closed 2022-06-18 by cezanne