cezanne/usbip-win #299: Does this client work with @Dorssel's USBIP Device Host server?

e-t-l, 2022-03-04:
The setup for the server in this readme seems pretty complex. A similar-looking project has their server setup packaged in an MSI installer: https://github.com/dorssel/usbipd-win.

Does anyone know, if you install Dorssel's USBIPD server on one computer and connect a USB device to it, will that be accessible from a device running the client from this repo?

e-t-l, 2022-03-04:
Confirmed with the folks at that repo that yes, it does:
https://github.com/dorssel/usbipd-win/issues/79#issuecomment-1059394012

stevogray, 2022-03-04:
Both packages are open source and readily available - you should test it for yourself using the specific USB device you want to use.
I have successfully used usbip-win in a VM and usbipd-win as the server with a few USB devices.

closed 2022-03-04 by e-t-l

sowri1, 2022-04-20:
Hi folks, I tried connecting to the drossel usbipd-win server using cezanne client. Although, i am able to view list of devices, when i try to attach the usb, it gets disconnected for some reason. I am attaching the log for reference.
```
info: UsbIpServer.ConnectedClient[1]
      Client 192.168.0.180 claimed device at 1-5 (USB\VID_0781&PID_5567\03025728080220025327).
info: UsbIpServer.ConnectedClient[2]
      Client 192.168.0.180 released device at 1-5 (USB\VID_0781&PID_5567\03025728080220025327).
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.IOException: Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host..
       ---> System.Net.Sockets.SocketException (10054): An existing connection was forcibly closed by the remote host.
         --- End of inner exception stack trace ---
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError , CancellationToken )
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource<System.Int32>.GetResult(Int16 )
         at UsbIpServer.Tools.ReadExactlyAsync(Stream , Memory`1 , CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Tools.cs:line 28
         at UsbIpServer.Interop.UsbIp.ReadUsbIpHeaderAsync(Stream , CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Interop\UsbIp.cs:line 181
         at UsbIpServer.AttachedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 510
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 216
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 239
```

Can you please help me with setting this client up?