cezanne/usbip-win #303: Easy setup?

Porkechebure, 2022-04-06:
Hello,
The setup for windows compiled version is hell. I checked readme.MD and I want to blind myself.
Luckily enough the author was kind enough to give a precompiled binary of the whole stuff. Lot of open source people don't do that and they should be whipped with some electrical wire peeled off and connected to the power line.
I have vs installed and I was already figuring the whole process of installing the whole c++ package, the whole windows driver kit and all the other stuff just to compile this.
Damn, I'm even a programmer myself and 15 years in the business and I'm finding complicated to set up all this stuff while some commercial counterparts offer a "install and go" with minor configurations
I'm reading that I have to set up a linux server specifically for this.... WHAT?

> Prepare a Linux machine as a USB/IP client or Windows usbip-win VHCI client

What? Why?
I tried some commercial options like "VirtualHere" and it does the server by itself and you just run it.

Why this has 3000 steps?
Can someone make a video to show how to setup this?

Thanks

fO-000, 2022-04-09:
The truth is you can download the precompiled binary on the release page, like [usbip-win-0.3.5.zip](https://github.com/cezanne/usbip-win/releases/download/v0.3.5/usbip-win-0.3.5.zip), and it contains `usbip.exe`, `usbipd.exe`, etc. The `README.md` actually states in a [not-so-obvious location](https://github.com/cezanne/usbip-win#windows-usbip-server) that the precompiled binary can be downloaded directly from the release page:

> You can find `usbip.exe`, `usbipd.exe`, `usbip_stub.sys` in the output folder after build or on [release](https://github.com/cezanne/usbip-win/releases) page.

I also think the author could put this useful information in a more visible place.

Porkechebure, 2022-04-10:
> The truth is you can download the precompiled binary on the release page, like [usbip-win-0.3.5.zip](https://github.com/cezanne/usbip-win/releases/download/v0.3.5/usbip-win-0.3.5.zip), and it [...]

Yes, as I said, I found the release links in some discussion and really didn't notice it in the guide (your link).

Yet, this seems a lot of complicated work to set up. I will try in the next days and let you know

vadimgrn, 2022-05-15:
> Hello, The setup for windows compiled version is hell. I checked readme.MD and I want to blind myself. Luckily enough the author was kind enough to give a precompiled binary of the whole stuff. Lot of open source people don't do that and they should be whipped with some electrical wire peeled off and connected to the power line. I have vs installed and I was already figuring the whole process of installing the whole c++ package, the whole windows driver kit and all the other stuff just to compile this. Damn, I'm even a programmer myself and 15 years in the business and I'm finding complicated to set up all this stuff while some commercial counterparts offer a "install and go" with minor configurations I'm reading that I have to set up a linux server specifically for this.... WHAT?
> 
> > Prepare a Linux machine as a USB/IP client or Windows usbip-win VHCI client
> 
> What? Why? I tried some commercial options like "VirtualHere" and it does the server by itself and you just run it.
> 
> Why this has 3000 steps? Can someone make a video to show how to setup this?
> 
> Thanks
> 
> OK, hidden here in the discussions I **_casually_** found this: https://github.com/vadimgrn/usbip-win2/releases
> 
> the motivation why these are hidden instead of being clearly advertised in readme.MD is beyond my understanding.
> 
> That said, can you be more clear about the working of these? Which one is server? Which one is client? Why it's usbip-win2? What's different from usbip-win?

Regarding usbip-win2, its README.md describes differences with original project. In two words, WDM vhci driver is significantly improved, UDE vhci driver is removed, stub driver (server) is used as is without modifications.

vadimgrn, 2022-05-28:
https://github.com/vadimgrn/usbip-win2/tree/develop

HybridAssassin, 2022-10-02:
I don't do programming but I'm really just trying to figure out how to redirect my second and third mouse to a second and third computer so, then I can play some emulators with friends using mouse mux. I read stuff normally and I can find out what to do. But I'm downloading stuff and I don't know what to do with it. I got the SDK build but I don't know how to use it. Not sure if that's to run it so the server can work?

The CMD pops out when I open usbipd.exe I assume that's for client side. I want to be the server so my friends can grab mouse's and we can play. I already installed the usbip_test.pfx. Though I just double clicked it and followed through made sure it was for local and not current user.
I have Windows 10. 

If it needs linux I gotta use a virtual machine.

HybridAssassin, 2022-10-02:
Umm also I put this usbip-win2-0.9.3.0-Release.exe in virusware Elect says it's a trojan.

vadimgrn, 2022-10-02:
This is false positive.

Kogotoro, 2022-11-02:
[Video_2022-10-25_171404.webm](https://user-images.githubusercontent.com/108133274/199440107-b7ac3b5a-c71a-48a3-9e4d-f1e40551e082.webm)

maybe someday... 
on video localhost test for usb-key with some certificate what popup install window for it on connect;
(not on video) it also was tested a bit on net connection for copy/create/delete/edit data on usb-storage, and using usb-video-cam.

-fixed some warnings/errors (mainly typecasts and not freed mem);

-added missing data for speeds;

-no 'on fly' generation/signing stub driver ini/cab files, its all prebuilded, but now its 3 versions(added auto detect/select for it): 
win8(without client-driver-ude-version), win10b1940-, win10b1941+(including win11) .
but win8 version still works for win10+ (dont remember checked or not w8.build  for w11), 
main cause deprecated ExAllocatePoolWithTag, what have 3 different versions what differently named and have different args types, and one of them also need addional call for some api on driver init -_-... it probably possible make one, but it need dynamicly load and resolve lib with ExAllocatePoolWithTag function.
(didn't try build for win7... code have some winapi for win8+.., starting from winsock2 and so on);

-no 'manual'(in code) pre-install stub-driver into storage, and select it from it by class or devid after, ie direct install from ini for device (it still installs into storage thought but windows do it);

-some usbip code moved to libusbip for using it also in usbip-gui;

-some usbipd code moved to libusbipd for using it also in usbip-svc;

-added service variant of usbipd (very raw based on msdn example =))

-added gui variant of usbip, what also manage service.

...so it possible fix existing code without (too) global rewrites =)