cezanne/usbip-win #307: The write operation fails in an Arduino device due to the network latency

myjimmy, 2022-04-26:
Hello,

I connected an Arduino device (Arduino Uno) on my PC (Windows 10) to the Windows Server 2019 machine by using the this project.
When the network latency is less than 40 ms,
1. I can mount the local Arduino device to the remote machine (Windows Server 2019).
2. **Arduino 1.8.19** app recognizes the Arduino device on the remote machine.
3. **Arduino 1.8.19** app reads the device info of the Arduino device on the remote machine.
4. I can write any sketch in the Arudino device by using **Arduino 1.8.19**.

But, wehn the network latency is more than 40 ms,
1. same to the above result.
2. same to the above result.
3. same to the above result.
4. When trying to write any sketch, the write operation always fails.
[image]

I need a solution to resolve this problem when the network latency is 60-100ms.

Thanks,
Jimmy

myjimmy, 2022-04-27:
@cezanne Do you have a good solution?

UnKnoWn-Consortium, 2022-05-07:
You can try setting a longer timeout at the avrdude config side of things. Tweak it till it works.

myjimmy, 2022-05-09:
@UnKnoWn-Consortium Thanks for your reply.
I'm a beginer to Arduino. I'm using `arduino-1.8.19-windows` (portable) on Windows 10.
I found the `avrdude.conf` file in arduino-1.8.19\hardware\tools\avr\etc.
But, because there are many items related to the timeout in the configuration file, I can't know which item I'll change.
So, I want you to let me know which item I'll change in the `avrdude.conf` file.

myjimmy, 2022-05-24:
My Arduino board is "Uno R3".
[image]
**Anything isn't connected to the Tx/Rx pins on my Arduino board.**

Also, I've already enabled the verbose mode in the Aruino IDE.
I upload the verbose log file here:
[arduino-cc-2.log](https://github.com/cezanne/usbip-win/files/8766135/arduino-cc-2.log)

UnKnoWn-Consortium, 2022-05-26:
It seems the toggle only adds one layer of verbosity and it does not give anything of value where the process hanged... 

Please give this modified command a try after ensuring the hex file exists or change its path to a hex file compiled from your sketch: 

```C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude.exe -v -v -v -v -C C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf -p atmega328 -c arduino -P COM3 -b 115200 -D -U flash:w:C:\Users\TESTUS~3\AppData\Local\Temp\3\arduino_build_672433\sketch_may18a.ino.hex:i```